opengeospatial/sensorthings #2: Temporal Logick

hylkevds, 2016-07-26:
For one of our projects we're using predictions, where each observation has a validTime interval. I just noticed the SensorThings API doesn't define any functions to query time intervals.

The most common query here would be "give me all observations that are valid at this time point". Something like 
`v1.0/Observations?$filter=during(2016-07-26T09:56:00Z, validTime)`
or
`v1.0/Observations?$filter=contains(validTime, 2016-07-26T09:56:00Z)`
But the best I can think of with the current set is
`v1.0/Observations?$filter=not validTime gt 2016-07-26T09:56:00Z and not validTime lt 2016-07-26T09:56:00Z `

But there are several more functions that would be very useful:

- before
- after
- starts
- finishes
- during/within/contains
- meets
- overlaps

hylkevds, 2016-10-12:
I've updated the suggested changes a bit:
https://github.com/hylkevds/SensorThingsAPI/compare/3e5bcc9c662ae24e6054d4823406c3ba83de35ae...master
and also created tests for the conformance test suite:
https://github.com/hylkevds/ets-sta10/blob/libraryTests/src/main/java/org/opengis/cite/sta10/filteringExtension/DateTimeTests.java

hylkevds, 2017-03-24:
I've grouped the changes for this issue in a single branch:
https://github.com/hylkevds/SensorThingsAPI/commits/timeFunctions

hylkevds, 2017-08-17:
A diff of the change is here:
https://github.com/hylkevds/SensorThingsAPI/compare/reworkRoot...timeFunctions

liangsteve, 2017-08-17:
@hylkevds let's use the same operations and definitions of OGC Filter Encoding's temporal operator. OGC 09-026r1 Filtering Encoding.

hylkevds, 2017-08-22:
Hmm, OGC 09-026r1 Filtering Encoding points to iso 19108 temporal operators, which is not free.
Assuming GeoTools implements the filters correctly, there are a few differences:

OGC 09-026R1 | iso 19108 | STA proposal | comment | geotools link
------------ | --------- | ------------ | ------- | -------------
After | After | after |  | 	[After](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/After.html)
Before | Before | before |  | [Before](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/Before.html)
Begins | Begins | starts | Proposal is wider | [Begins](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/Begins.html)
BegunBy | Begun | - | | 
TContains | Contains | - |  |
During | During | during | iso: BothOpen. Proposal: half-open | [During](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/During.html)
Ends | Ends | finishes | proposal is wider | [Ends](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/Ends.html)
TEquals | Equals | equals |  | [TEquals](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/TEquals.html)
TOverlaps | Overlaps | overlaps | proposal: also for instants | [TOverlaps](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/TOverlaps.html)
Meets | Meets | meets | iso: 1<2. Sta: both ways, also instants | [Meets](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/Meets.html)
OverlappedBy | OverlappedBy | - |  |
MetBy | MetBy | - |  |
EndedBy | EndedBy | -	 |  | 
AnyInteracts | N/A | -  | same as overlaps in proposal | [AnyInteracts](http://docs.geotools.org/latest/javadocs/org/opengis/filter/temporal/AnyInteracts.html)

First of all, the proposal leaves out the inverse operations. I find they just clutter the standard.

Begins / ends: For two intervals, iso 19108 requires the first interval to be shorter than the second. In our proposal, the length of the intervals doesn't matter, as long as they start/end at the same time. The proposal uses the names "starts/finishes" but that is a small change.

During: iso 19108 uses both-open intervals. This means that if we take a series of  contiguous intervals (t1-t2, t2-t3, t3-t4, t4-t5) then iso 19108 defines t1, t2, t3, t4 and t5 not to be "during" any of these intervals. I think that is highly unexpected for users. Therefore the proposal defines intervals to be half-open, with the starting time being part of the interval, but the end-time not.

Overlaps: iso 19108 defines overlaps only for intervals. The proposal also allows instants.

Meets: iso 19108 defines that the first of the two intervals has to come first (t1.end = t2.start). The proposal also allows the other way around.

What do you all think?

liangsteve, 2017-09-11:
Note from STA SWG at Southhapton TC: ask Chris Little and/or Simon Cox regarding ISO time standards. Especially on why overlaps is both open.  Seems counter-intuitive.

Will vote in SWG after asking the experts.

hylkevds, 2018-03-18:
I think ISO defines times to always be intervals, meaning that a time "instant" actually has a duration, and the given time instant is actually the start of a very short interval.

If that's so, then the result is that intervals are half-open, with the start being part of the interval. A time instant at a boundary between two contiguous intervals will overlap with the second interval.

In that case, an info box explaining the half-open behavior would be sufficient.

hylkevds, 2018-10-15:
In the pull request I've added the time functions as defined in OGC 09-026r1 Filtering Encoding

humaidkidwai, 2024-12-07:
Why was PR #54 closed? Are we still thinking about including this for v2.0?

hylkevds, 2024-12-08:
Good question.
#54 was closed because adding functionality was deemed to not be a good idea for v1.1

We could add it to V2, though it may not be as necessary, since v2 allows filtering on the `start` and `end` of intervals. It may still be good as convenience though...

hylkevds, 2025-03-12:
We already have similar functions named st_*, like st_overlaps...
Since `st` stands for spacial-temporal, should these not also work for times?