opengeospatial/sensorthings #3: Receive updates with MQTT Subscribe

mjacoby, 2016-07-27:
Given the following scenario

**Existing entities:**

- Sensors(1) 
- Sensors(2)
- Datastreams(1)

Datastreams(1) and Sensors(1) are linked, therefore 
`Datastreams(1).Sensor = Sensors(1) and Sensors(1).Datastreams = Datastreams(1)`

**Existing MQTT subscriptions:**

- /Sensors(1)
- /Sensors(2)
- /Datastreams(1)
- /Sensors(1)/Datastreams
- /Sensors(2)/Datastreams
- /Datastreams(1)/Sensor

Now Datastreams(1) is linked to Sensors(2), e.g. 
`Datastreams(1).Sensor = Sensors(1) --> Datastreams(1).Sensor = Sensors(2)`

**Which of the given MQTT subscriptions will get notified any why?**

mjacoby, 2016-08-03:
Some other questions regarding on receiving updates with MQTT subscribe I've came across

- Can the the URL and port of the MQTT server be chosen freely and if so, how do clients know where to find the MQTT server? Maybe this could also be included in some kind of capabilities document as mentioned [here](https://github.com/opengeospatial/sensorthings/issues/4).
- What exactly is published via MQTT when an entity is update? In the standard it says in Section 14.2.2

> Response: When a property of the subscribed entity is updated, the service returns a complete JSON
representation of the updated entity

but there is no definition of what _a complete JSON representation_ is. Does it include the selflink and all navigationLinks?

- If navigationLinks should be included in the response I suppose they must be absolute because a relative URL inside an MQTT message is meaningless as the base URL to resolve the relative URL is not available.

- Is there any assumption how the MQTT server should react to subscriptions to invalid topics like **Thing/Datastreams/Observations**? Should they maybe be rejected on a MQTT level or accepted on MQTT level but nothing ever will be published on them (at least from the server)?

liangsteve, 2017-08-17:
See https://portal.opengeospatial.org/files/?artifact_id=70669 for an update. Removed MQTT Subscribe ENTITY/# (e.g., Datastreams(1)/#).  We believe it is clear if a client specify clearly what does it want to receive updates about (e.g., Datastreams(1)/update, Datastreams(1)/delete, etc.)