opengeospatial/sensorthings #7: Issues with datatype Any

mjacoby, 2016-09-08:
_This issue has already been submitted as change request and is just posted here to have a central place where all the issues can be found_

SensorThings API introduces the datatype _Any_ for some entity properties (namely Sensor.metadata, Location.location, Observation.result, FeatureOfInterest.feature) which can take any value specified in JSON. This raises two issues:

- How are these values stored? Should the JSON datatype be preserved when inserting a property of datatype Any (e.g. when inserted as JSON Number can it be expected that it will be stored and returned as JSON Number)?
- Semantics of some operators are undefined when used on properties of type _Any_
    - $orderby clauses on properties of type _Any_
    - built-in filter operation using properties of type _Any_ as operand
    - built-in query function using properties of type _Any_ as parameter

For the detailed description please see the [change request 383](http://ogc.standardstracker.org/show_request.cgi?id=383)

hylkevds, 2016-09-20:
Here is a set of diffs to the current document with the suggested changes:
https://github.com/hylkevds/SensorThingsAPI/commit/64187ada6b7fd36a0351a7ee023cdbc5290c01a0
https://github.com/hylkevds/SensorThingsAPI/commit/ccae6b261ca72dd2b7137d7a3d40c1bc340874a7
https://github.com/hylkevds/SensorThingsAPI/commit/3e5bcc9c662ae24e6054d4823406c3ba83de35ae

hylkevds, 2017-03-24:
I've reworked the changes for this issue into a specific branch:
https://github.com/hylkevds/SensorThingsAPI/commits/dataTypes