opengeospatial/sensorthings #9: Datastream.observedArea for a single point

hylkevds, 2016-09-29:
Hi All,

Datastream.observedArea is defined as GM_Envelope (GeoJSON Polygon)
What to do when the features of interest for a given Datastream are just a single point?

1. Return a Point (though the standard states it should be a polygon)?
2. Return a polygon containing just 4x that point as corners?
3. Return nothing?

What's the general opinion on this?

liangsteve, 2016-09-30:
Anyone knows how does typical OGC capabilities XML handle this?

dpmcgarry, 2017-12-07:
+1 on this issue as it somewhat relates to the issues in #33

hylkevds, 2019-07-08:
We could change the definition for v1.1 to allow non-polygons (any GeoJSON object) for the Datastream.observedArea. Or would there be issues with that?

nmtoken, 2025-03-12:
FROST is currently allowing a point e.g.

    {
      "@iot.selfLink": "https://sensors.bgs.ac.uk/FROST-Server/v1.1/Datastreams(23)",
      "@iot.id": 23,
      "name": "GGERFS_01 air_temperature",
      "description": "air_temperature from BGS GGERFS Barometer - Measured (C)",
      "observationType": "OM_Measurement",
      "unitOfMeasurement": {
        "name": "Degrees Celsius",
        "symbol": "C",
        "definition": "https://vocab.nerc.ac.uk/collection/P06/current/UPAA"
      },
      "observedArea": {
        "type": "Point",
        "coordinates": [-4.201199802, 55.838255175]
      },
      "phenomenonTime": "2020-03-16T14:02:18Z/2025-01-08T13:00:00Z",
      "properties": {
        "access_restriction": "Unrestricted",
        "data_type": "Measured",
        "data_usage": "Available under the Open Government Licence subject to the following acknowledgement accompanying the reproduced UKRI materials \"Contains UKRI materials ©UKRI [year]\"",
        "publish_yn": "Y",
        "sen_id": 827
      },
      "resultTime": "2020-03-16T14:02:18Z/2025-01-08T13:00:00Z",
      "ObservedProperty@iot.navigationLink": "https://sensors.bgs.ac.uk/FROST-Server/v1.1/Datastreams(23)/ObservedProperty",
      "Sensor@iot.navigationLink": "https://sensors.bgs.ac.uk/FROST-Server/v1.1/Datastreams(23)/Sensor",
      "Thing@iot.navigationLink": "https://sensors.bgs.ac.uk/FROST-Server/v1.1/Datastreams(23)/Thing",
      "Observations@iot.navigationLink": "https://sensors.bgs.ac.uk/FROST-Server/v1.1/Datastreams(23)/Observations"
    }

nmtoken, 2025-03-12:
> Anyone knows how does typical OGC capabilities XML handle this?

WCS 2.0.1 doesn't give any extent in the GetCapabilities, though isn't typical usage.  However `BoundingBox` and `WGS84BoundingBox` used in OWS GetCapabilities are imported from `OGC Web Services Common Specification` and are specified as `0.. *`