opengeospatial/sensorthings #10: Remove tests for PUT from Conformance Classes

taniakhalafbeigi, 2016-10-04:
Since PUT is an optional functionality in SensorThings, there is no need for testing it in conformance test suite.

hylkevds, 2018-11-27:
You've closed this issue, but not referenced any commit or pull request, and the "problem" still exists. Did you forget to push any changes?

taniakhalafbeigi, 2018-11-27:
I opened the issue for the test suite and not the spec in the first place and closed this as of that.
But you are right we need the change it in the spec as well. I will reopen it and thanks for the PR.

rhzg, 2018-11-28:
I believe, if an implementation of the SensorThings API is implementing a optional feature, it must implement it correctly. Otherwise it shall not be implemented at all. And that should be part of the conformance test. Otherwise optional features run the risk of being implemented in incompatible ways.

KathiSchleidt, 2018-11-28:
Agreed, either its clearly specified and testable, or it's NOT part of the standard

liangsteve, 2018-11-30:
Great discussions.  Let's not change the STA specification.  I think any changes to the specification needs to be treated very seriously.  When the SWG designed STA, the text about PUT is really a warning.  It meant to say we SHALL use PUT correctly.  I think we should test it.  Let's keep it the way it is, _i.e._, no change to the specification and the test suite implementation should be enhanced. 

e.g., 
(1) Check PUT supported by a server or not. (e.g., a server SHALL return 501 Not Implemented if PUT is not supported)
(2) If PUT is supported by the server implementation, then it should be implemented correctly.

Hope this is clear.  Thank you all.

commit a2fd2530 (2020-09-14): Fixes #10, specify how a server must respond when PUT is not implemented
PR #59 (2020-09-14): 18-088: Fixes #10, Specify put not implemented
This fixes #10 by specifying how a server shall respond when it does not implement PUT.
closed 2022-02-16 by hylkevds