opengeospatial/sensorthings #12: Filter results and duplicates

hylkevds, 2016-10-06:
Hi all,

Another opinion question. When doing deep filters, like 
`v1.0/Things?$filter=startswith(HistoricalLocations/Location/name, 'Location 1')`
there might be duplicates in the result set: A thing might have been at 'Location 1' more than once, or a thing might have been at 'Location 1' and 'Location 10'.

What do you all think: Should duplicates be returned, or should each thing appear in the result only once?

taniakhalafbeigi, 2016-10-06:
I think this query does not need duplicate results. 
It means return Things that was in Locations with the name starts with 'Location 1' at some time. So you just return each of the Things with that condition once (I mean it doesn't matter if that Thing was in that Location multiple times or in different Locations that satisfies the filter). If you expand HistoricalLocation/Location, then the list of HistoricalLocation for each Thing has entities with Location name starts with 'Location 1' and this list is never empty for this query.
I know that considering it as SQL with the joins we may have a Thing returned multiple times for each of the Locations, but in showing the result I believe we shouldn't show the duplicates.
Do you see any advantage in showing duplicate result?

hylkevds, 2016-10-07:
I also do not see any advantages in returning duplicates, so our implementation does filter those out.

I ran into the issue while writing test for the functions, so I'm interested to see how other people think about the issue.

I doubt it will ever cause problems if different implementations behave differently here, so testing for it in the conformance tests is probably not needed.

closed 2018-12-27 by taniakhalafbeigi