opengeospatial/sensorthings #16: $expand with multiple expandItems

hylkevds, 2016-12-15:
I was looking into writing some tests for nested expands when I noticed the following:

SensorThings API paragraph [9.3.2.1 $expand](http://docs.opengeospatial.org/is/15-078r6/15-078r6.html#47) refers to [OData 4.0-URL 5.1.2](http://docs.oasis-open.org/odata/odata/v4.0/errata03/os/complete/part2-url-conventions/odata-v4.0-errata03-os-part2-url-conventions-complete.html#_Toc453752359), which states:

> A navigation property MUST NOT appear in more than one expandItem.

The conformance test generates urls like:
`/v1.0/Things?$expand=Datastreams/Thing,Datastreams/Sensor`

Doesn't the navigation property Datastreams of the entity type Thing appear in more than one expandItem here? I would think that should be rewritten as
`/v1.0/Things?$expand=Datastreams($expand=Thing,Sensor)`

(The OData [ABNF](http://docs.oasis-open.org/odata/odata/v4.0/errata03/os/complete/abnf/odata-abnf-construction-rules.txt) defines navigationProperty as odataIdentifier which means that Datastreams/Thing consists of two navigationProperties)

lieberjosh, 2019-06-05:
Not sure this is a problem with OData (OData 4.0-URL 5.1.2):

"An arbitrary number of single- or collection-valued complex properties, optionally followed by a type cast, allow drilling into complex properties.
...
Example 92: expand a navigation property of a complex type
http://host/service/Customers?$expand=Addresses/Country"

So Addresses/Country is a distinct, if complex, navigation property, and a list of them should be conformant.

hylkevds, 2022-10-17:
The navigation property in that example is `Country` which is a property of the complex type of the `Addresses` property of the `Customer` Entity. As such, that navigation property `Country` is only allowed to appear in one expand item. 
So if you wanted to expand `Addresses/Country/Capital` and `Addresses/Country/Cities` you'd have to write `?$expand=Addresses/Country($expand=Capital,Cities)` 
and not 
`?$expand=Addresses/Country/Capital,Addresses/Country/Cities`

The reasoning behind this is probably that you can add options to the expanding of navigation properties, and if a navigation property can appear multiple times, the options can differ and the result would be undefined.