opengeospatial/sensorthings #23: URI to opengis are not working

geoxanadu, 2017-05-16:
The external identifiers, i.e. http://www.opengis.net/doc/is/sensorthings/1.0 in the SensorThings document (http://docs.opengeospatial.org/is/15-078r6/15-078r6.html) are not working anymore.

dpmcgarry, 2017-12-07:
+1 we are seeing the same issue

KathiSchleidt, 2019-07-17:
Can we have @ghobona look at it

ghobona, 2019-07-17:
Thanks for spotting the issue.

I have logged the issue with the OGC Infrastructure Office. They will let us know when it is fixed.

ghobona, 2022-08-04:
The external identifiers are now resolving, so the problem has been fixed.

This GitHub Issue can be closed.

hylkevds, 2023-08-28:
For 1.1 the URLs are also not working. Everything below
`http://www.opengis.net/spec/iot_sensing/1.1/`
Should probably redirect to 
`https://docs.ogc.org/is/18-088/18-088.html`

hylkevds, 2023-08-28:
@ghobona, can you have another look at it?

kstegemoller, 2023-08-28:
Regarding ... http://www.opengis.net/spec/iot_sensing/1.1/ => https://docs.ogc.org/is/18-088/18-088.html

It looks like 18-088 was not submitted to the Naming Authority.  There are no "iot_sensing/1.1/" mappings in the NA repo https://github.com/opengeospatial/NamingAuthority.

Attached
[18-088.csv](https://github.com/opengeospatial/sensorthings/files/12458359/18-088.csv)
 is a first attempt at the mappings.

hylkevds, 2023-08-31:
That mapping looks good.
Would that also redirect partial paths? A server implementing the entire data model only needs to advertise `http://www.opengis.net/spec/iot_sensing/1.1/req/datamodel`, so partial paths starting with `http://www.opengis.net/spec/iot_sensing/1.1/req` should just redirect to the document itself.

kstegemoller, 2023-08-31:
@hylkevds I have updated the CSV to include the parent paths.  
See it at https://github.com/opengeospatial/NamingAuthority/blob/master/specification-elements/mappings/18-088.csv

I will make them live later today.

kstegemoller, 2023-08-31:
@hylkevds The iot_sensing/1.1 URI's are now active.  Thank you for you help.

closed 2023-09-05 by hylkevds