opengeospatial/sensorthings #24: Version prefix in MQTT topics

hylkevds, 2017-06-07:
This point was raised in the test suit pull request: https://github.com/opengeospatial/ets-sta10/pull/20

In the HTTP requests, all URLs contain the version prefix (v1.0). The MQTT topics do not contain this prefix. This means the MQTT topics are not "version safe". Was this on purpose, or should we add the version prefix to the MQTT topics as well?

hylkevds, 2019-04-23:
What do current implementations do?
FROST wants the version prefix.
How about SensorUP and Gost?

taniakhalafbeigi, 2019-04-23:
We have version prefix like `v1.0/Observations`.

bertt, 2019-04-24:
In GOST there is no version info in the topic (just like the samples in the spec pdf)

hylkevds, 2022-02-23:
Fixed with the publication of Sensing v1.1.

commit 3c2d3400 (2020-09-14): Fixes #24: version prefix in MQTT topics
closed 2022-02-23 by hylkevds