opengeospatial/sensorthings #25: What is the unit of distance method in sensorthing API

sudhamani1983, 2017-07-22:
Please help provide unit details of geo.distance method.

hylkevds, 2017-07-24:
They are in the same units as the coordinates of the coordinate system you use, or more accurately, the coordinate system that the server uses to store the geometries.

Since the default coordinate system of GeoJSON is "WGS 84, with longitude and latitude units of decimal degrees", the result of geo.distance will usually also be in decimal degrees too.

selimnairb, 2019-04-23:
This seems to be complicated enough to be revisited in v2.0.  For example, GeoJSON requires WGS84, but implementations are free to store data in any coordinate system they want.  Also, users may want to specify the coordinate system to use for queries.  For example, I may want to query for distance in meters from a UTM easting and northing.  In this case, I would rely on the server to do the transform for me.  This is more user friendly than the current form of requiring all queries to be done in WGS84 degrees.  

At the very least, it seems like an oversight that an OGC standard does not explicitly represent (i.e. in  a programmatically discoverable way at runtime) the coordinate system used for geographic data.  Perhaps for v1.1 we can add section to the [proposed server settings metadata](https://github.com/opengeospatial/sensorthings/issues/4) documenting the coordinate system.  This will clarify the immediate issue, while providing a foundation for potentially dealing with this more comprehensively in v2.0.

Thoughts?