opengeospatial/sensorthings #28: Implementing Security on all Entity access (Things, Datastreams, Observations etc.) based on User Access on Things

chhonker, 2017-09-27:
All,

I have data from multiple platforms (treated as "Things" ). The Thing type is defined in Thing->properties {platformType:Drifting} (The platform type can be one out of many platforms I have).

I will create a Users_Roles Table where every user will be given access to "0 or more" Things platformType.

I want that a user can access _Datastreams_, _Observation_ only for the _Things_ platformType it has Access.

* e.g. for Things I've tried (/sensor/v1.0/Things?$filter=properties/type eq 'drifting') but this required a lot of regex matching in the backend to extract filter values and allow.

How should I secure Observations and Datastreams? Is there any internal support from OGC SensorThings API for this? Or can someone suggest any project related to this.

hylkevds, 2017-09-27:
The SensorThings API itself does not address security at all.

Data-based security is one of the things we have on our ToDo-list for our Fraunhofer SensorThingsServer, but it might be a while before we get to it.

You're right that that is almost impossible to do with an external security solution, because there are many paths that lead to the same data. Our idea is to add this directly to the database back-end and modify the SQL query we generate to also include the security check. How to do this in a generic way is the hard part.

chhonker, 2017-09-27:
Thanks @hylkevds 
I'm actually using the Fraunhofer STS implementation and will wait for the security code check-in.

hylkevds, 2024-07-29:
FROST can now do such fine-grained authorisation. An example of how this can be used can be found in the [Projects Plugin](https://github.com/hylkevds/FROST-Server.Plugin.Projects/).