opengeospatial/sensorthings #31: Absolute references in filters

hylkevds, 2017-11-03:
Is it allowed to use absolute references as filter parameters? 
For example: I have a Location with id 5, that is a polygon, can I say 

> give me all FeaturesOfInterest that are within Location 5

using the query: 
```v1.0/FeaturesOfInterest?$filter=st_within(feature, /Locations(5)/location)```

Since `/Locations(5)/location` resolves to a single geometry, this would be a nice way to get around the very verbose urls you get for spatial queries.

KathiSchleidt, 2020-11-07:
Hiya,
anything happening on this issue? Would be really valuable!
Also - while we're at it, would be nice if the geometry could be valid geoJSON at the end of any reference - that way we could provide a reference to an OGC API - Features feature as providing the query geometry!
:? 
Kathi

hylkevds, 2024-07-29:
OData does this with the [$root keyword](https://docs.oasis-open.org/odata/odata/v4.01/os/part2-url-conventions/odata-v4.01-os-part2-url-conventions.html#sec_root):

> 5.1.1.14.5 [$root](https://docs.oasis-open.org/odata/odata/v4.01/os/part2-url-conventions/odata-v4.01-os-part2-url-conventions.html#sec_root)

> The $root literal can be used in expressions to refer to resources of the same service. It can be used as a single-valued expression or within [complex or collection literals](https://docs.oasis-open.org/odata/odata/v4.01/os/part2-url-conventions/odata-v4.01-os-part2-url-conventions.html#sec_ComplexandCollectionLiterals).

> Example 108: all employees with the same last name as employee A1235
> `http://host/service/Employees?$filter=LastName eq $root/Employees('A1245')/LastName`