opengeospatial/sensorthings #32: Using SWE Common/SensorML encodings for observations

mikebotts, 2017-12-04:
Recently, the models for SWE Common Data and SensorML were used to create a JSON encoding for these standards. These JSON encodings are currently being used to support

- tasking parameters in STA
- SensorML and observation templates and dynamic data streams in OpenSensorHub

It is highly recommended that we look at using the SWE Common/SensorML encodings to support observations (as well as tasking) and MQTT message properties within the STA standard. Use of the SWE Common/SensorML models and JSON encoding for both tasking and observations provides the benefits of:

- balancing the tasking and measurement models in STA (as is available in SWE)
- providing closer compatibility between STA and SWE
- makes use of well tested and robust data models in STA

Thanks.

hylkevds, 2017-12-05:
Could you give some concrete examples of how things would change?

Sam-Bolling, 2017-12-05:
For defense, intelligence, military, and government applications, does this effort involve, or have implications for the use of, AMQP; potentially in lieu of MQTT from a security perspective?

mikebotts, 2017-12-05:
Good question. I know STA is built upon or relies on MQTT, at least in the implementation, but I don't know if the communication protocol is limited to MQTT in the standard. This would be the case for both observations and tasking I assume. Steve?

To be clear (I think), the request for SWE Common use here only affects the models and encoding within the messages and shouldn't affect the message exchange protocol.

mikebotts, 2017-12-05:
@hylkevds Some examples of using the new JSON encoding within SWE can be found at the bottom of demo links at http://sensiasoft.net:8181/demo.html  (JSON Support)

mikebotts, 2017-12-05:
I should note for clarification that the JSON encodings were generated from the same models that were used to generate the original XML encodings of SensorML and SWE Common Data.

Sam-Bolling, 2017-12-07:
@mikebotts Rereading the original suggestion from your initial post for STA to use the newly developed, SWE Common/SensorML-based JSON encodings, what does this look like for STA moving forward?  Does STA already have JSON encodings in use?  Would these be replaced or augmented somehow?

matthijskooijman, 2025-04-19:
FWIW, the current SensorThings 2.0 spec that is being written in this repo seems to incorporate this suggestion, using SWE common for defining a Datastream resultType and Observation result: https://github.com/opengeospatial/sensorthings/blob/23-019/sections/clause_07_core_data_model.adoc#datastream