opengeospatial/sensorthings #36: Use of parenthesis in URL

dpmcgarry, 2017-12-07:
The spec addresses specific items using (X) in the URL. This seems to be at odds with the "traditional" REST API approach of having /{category}/{X} to address a specific item.  We have found the use of parenthesis to be incompatible with API Gateways which is causing us to write specific workaround code to handle the use of parenthesis.

Request the following considerations:
1. Allow a URL with a parenthesis to respond with a HTTP 302 to a URL with a / delimited item
2. Permit the use of / delimited items if they are defined as such in the /v1.0/ base URL

taniakhalafbeigi, 2017-12-07:
SensorThings is following OData URL conventions, and that is the reason we use parenthesis instead of traditional REST.
For an implementation to be compliant to SensorThings API, it must implement the URL conventions as stated in the API. 
However, an implementation can implement the traditional REST as well as OData as an extra feature and still be compliant to SensorThings.

dpmcgarry, 2017-12-07:
Ok thanks for the information.  I had a feeling the (X) was an ODATA thing, but thank you for confirming.  In doing some research it appears others are requesting the API Gateways I mentioned to support OData, so we will continue with our workaround that supports both for now.  Thanks again!

hylkevds, 2018-11-16:
To elaborate a bit on this:
One of the reasons OData does this is to make it completely unambiguous which part of the path is an ID and which is a property. OData is a generic interface for any database schema, so you can have Strings for IDs, and numeric property names. It would be much more difficult to determine which of a/b/4/2/d are the ids and which are properties or navigation links.

closed 2018-11-16 by liangsteve