opengeospatial/sensorthings #38: Compatibility with the Semantic Sensor Network Ontology?

abourdon, 2017-12-15:
Hi,

Is there a goal to be compatible in the future with the OGC's [Semantic Sensor Network Ontology](https://www.w3.org/TR/vocab-ssn/)?

Regards,
Aurélien

lieberjosh, 2018-03-23:
Both are aligned with Observations and Measurements, as is SOS, but not identically. SensorThings Datastream does not exist in SOSA/SSN, but may align with the proposed [SOSA:ObservationCollection](https://w3c.github.io/sdw/proposals/ssn-extensions/#sosa:ObservationCollection). SensorThings Thing does not exist in SOSA/SSN and it is unclear how that would be aligned.

There are a number of parts of SOSA/SSN relating to sampling and actuating, as well as procedures, platforms, etc. that are not part of Sensor Things and await specific proposals as to how they might be incorporated.

lieberjosh, 2018-03-23:
Thinking about this further, it may be that the best alignment for a SensorThings "Thing" is with SOSA:Platform. This allows commonality of location for some number of Datastreams, Sensors, ObservableProperties, etc. and is relatively agnostic to whether it moves between observations as well as whether observations made from it target the same or different features of interest. Only slightly complicated by the fact that there is no direct connection as in SOSA between Sensor and Thing, only through Datastream. Perhaps SOSA: Platform should be added as a common property to ObservationCollection.

lieberjosh, 2018-04-19:
Another alignment issue is that not all [SOSA](https://www.w3.org/TR/vocab-ssn/images/SSN-Observation.png) relations go through Observation and the proposed SOSA extension sosa:ObservationCollection. A sos:Sensor observes a sosa:ObservableProperty. A sosa:ObservableProperty isPropertyOf sosa:FeatureOfInterest (through ssn:Property). sosa:Platform (isSameAs stapi:Thing ?) hosts sosa:Sensor. Granted that the relation chain through stapi:Datastream can be followed in STAPI, but it does require the existence of an appropriate Datastream in order to do so.

abourdon, 2018-04-26:
Thank you @lieberjosh for these remarks. Indeed, both are aligned with Observation perspective but using some minor differences. I'm looking for a way to inject data provided by a SensorThings API system to an SSN/SOSA based ontology.

Indeed, I think we could have these (approximatively) equivalences:
- stapi:Thing ~ sosa:Platform
- stapi:Location ~ ssn:forProperty of a ssn:Deployment
- stapi:HistoricalLocation ~ ssn:forProperty of a ssn:Deployment
- stapi:Sensor ~ sosa:Sensor
- stapi:ObservedProperty ~ sosa:ObservableProperty
- stapi:Observation ~ sosa:Observation
- stapi:FeatureOfInterest ~ sosa:FeatureOfInterest
- stapi:Datastream/stapi:MultiDatastream ~ sosa:ObservationCollection (in a future version of SOSA/SSN)

abourdon, 2018-04-26:
@lieberjosh, Even if a sosa:FeatureOfInterest could have a location, I don't know if conceptually a stapi:Thing can be aligned with a sosa:FeatureOfInterest. As wrote in the definition:

> Feature Of Interest - The thing whose property is being estimated or calculated in the course of an Observation to arrive at a Result, or whose property is being manipulated by an Actuator, or which is being sampled or transformed in an act of Sampling.

As for SensorThings API, a sosa:FeatureOfInterest is only the entity from which a sosa:Sensor wants to observe a particular sosa:ObservableProperty.

IMHO, to align with SensorThings API, notion of stapi:Location or stapi:HistoricalLocation should be directly correlated with a sosa:Platform. However, there is no similar property that could contain these locations. The best I could find was to use the ssn:isDeployment relation that give a ssn:Deployment from which we use the ssn:forProperty relation to point to a ssn:Property. And regarding the documentation, this ssn:Property could contain our locations:

> Property - A quality of an entity. An aspect of an entity that is intrinsic to and cannot exist without the entity.

What do you think?