opengeospatial/sensorthings #44: Extension: DELETE on collection, with $filter

hylkevds, 2018-03-20:
Data cleanup with the SensorThings API is currently very inconvenient. If you want to regularly delete all Observations older than 3 years, you have to first GET all those observations' ids, and then DELETE each Observation separately. You can do it in one Request using batch, but you still have to list each DELETE command.

I would suggest extending DELETE to also work on collections, and allow $filter to state which entities should be deleted.

Deleting all Observations older than 3 years could be as simple as:
```
DELETE v1.0/Observations?$filter=phenomenonTime lt now() sub duration'P3Y'
```

chhonker, 2018-08-24:
Is this functionality available in FROST-Server ?

hylkevds, 2018-08-24:
Not yet, it's on our TODO list.

hylkevds, 2018-08-27:
The changes mentioned in #3 also add delete notification to the standard. It's probably a good idea to allow a server to not send notifications for items deleted using a filter.