opengeospatial/sensorthings #46: Implement Procedure extension

lieberjosh, 2018-04-19:
Sensor output may need to be processed in some way to yield useful observations. The procedure for such processing is currently not visible or definable in SensorThings. An extension that added a procedure entity aligned with [sosa:Procedure](https://www.w3.org/TR/vocab-ssn/images/SSN-Observation.png) could allow the procedure behind a datastream to be visible or even definable via a service call or supported scripting API. Alternately, additional datastreams could be added for a given sensor to provide refined results, such a moving average or functions of multiple readings such as temperature-corrected altitude.

KathiSchleidt, 2018-11-16:
Hi Josh,
could you clarify what the Input/Output from SSN would/could be? Could this be a different datastream?
:?

KathiSchleidt, 2018-11-16:
Thanks Josh :)
for further clarification - could I state that a sta:Datastream isA ssn:Input for the next Procedure down the processing chain?

lieberjosh, 2018-11-16:
Now that gets a little tricky, because sosa:Procedure is a recipe, not a bakery. Might need to have some sort of an ssn:System, e.g. stapi:Process, that implements a Procedure which in turn has for inputs the datatypes of the Datastream elements of interest (e.g. result, time, observed area). Then one would configure that Process to draw from a particular Datastream.

lieberjosh, 2018-11-16:
> but that's where I appreciate STA's split of Datastream and Observations, to my view, the Datasteam becomes the ingredient description while the Observations are the actual ingredient
> :?

I don’t think it’s exactly a split. A datastream is a convenient collection of observations, not the definition of them. A procedure (e.g. relative humidity) may use temperature measurement as an input, but more than one datastream may collect termperature measurements.

hylkevds, 2018-11-17:
A sensor does not have to be modelled as an individual sensor. It can also be a sensor type. In that case you can store the procedure information with the sensor.
If you model each sensor you can also do that, but you'd get some duplication.

KathiSchleidt, 2018-11-17:
Wait - now we're back to the old argument of if the procedure/Sensor is the class or the instance
Under O&M I'd adjusted to the fact that it should be the class of measurement devices, any specific parameterization should be provided in the Observation parameter
Is this formally specified in STA, or just devolved from O&M?

lieberjosh, 2018-11-19:
There may be some metalevel ambiguities in O&M and SOSA/SSN, particularly with regard to ObservedProperty, but I don’t think this is one of them. Sensor is a class, as are any of the other concepts in the model / ontology. One defines a specific instance of the class to represent observation data. However, that instance may be generic, e.g. all such thermocouple products are reasonably indistinguishable, or it may be specific, e.g. each specific thermocouple has a unique calibration curve and service history. Specificity also affects traceability, e.g. if a Sensor instance is generic, it won’t be possible to ask which specific sensors are mounted on a specific platform. That’s a modeling choice, but not a metalevel issue.

It still leaves open what parameters are recorded in a Procedure, what in a Sensor (is a System), what in a putative Process (is also a System) and what in an Observation. The general principle would be to record them where they are likely to remain most constant, but there is definitely room for modeling choice here as well. STAPI makes some specific choices and not others.