opengeospatial/sensorthings #51: CSV result format - V1.1 Request

KathiSchleidt, 2018-07-02:
In order to make STA data more accessible, it would be valuable to provide a simple CSV result format

liangsteve, 2018-12-10:
I totally agree that it's very useful to have CSV outputs. Can you please offer more details here? There are many many different ways to flatten the data model and provide CSV outputs. Thoughts? (originally posted [here](https://github.com/INSIDE-information-systems/SensorThingsAPI/issues/3))

hylkevds, 2019-06-07:
Other issues dealing with alternative result formats are:
* #70 Add GeoJSON format for selected outputs
* #45 Add relation between FeatureofInterest and ObservedProperty (Get sosa aligned response)
* #32 Using SWE Common/SensorML encodings for observations

sgrellet, 2019-09-10:
Copy / Paste from the proposal made in the [original post ](https://github.com/INSIDE-information-systems/SensorThingsAPI/issues/3#issuecomment-501734881): 

"Proposal from our workshop at BRGM with Hylke and Kathi and BRGM folks
Rationale only traverse the single cardinalities
* mainly feasible for Observation and Datastream
  * Observation : start from Observation, traverse the single cardinalities (no funky loops) and write everything flat at the result level. Will only miss the Location
  * Datastream will have Thing, Sensor, ObservedProperty but no Observation nor Location
* for the others, its just a csv 'dump' of the single table
* MultiDatastream might a bit tricky"

hylkevds, 2020-02-05:
I made a write up for a CSV resultFormat extension at:
https://github.com/INSIDE-information-systems/SensorThingsAPI/blob/master/CSV-ResultFormat/CSV-ResultFormat.md

Comments are welcome, it's not implemented yet.

hylkevds, 2020-02-20:
I've implemented the CSV extension. If anybody want to try it out, here is a link to a demo service:
https://ogctest.docker01.ilt-dmz.iosb.fraunhofer.de/v1.1/Observations?$resultFormat=CSV&$select=id,phenomenonTime,result&$expand=Datastream($select=id,unitOfMeasurement;$expand=ObservedProperty($select=id,name))

mbeaufils, 2020-02-21:
Would it be possible to set other separators instead of ","? For example ";" or "TAB". The possible issue I anticipate is some looong description fields that would contain "," in it.

hylkevds, 2020-02-21:
[RFC4180](https://tools.ietf.org/html/rfc4180) describes how things like commas, quotes and line breaks should be handled in CSV files.
If you change the separator to anything else, you've just moved the problem to that other character, which is just as likely to be in the strings...

mbeaufils, 2020-02-21:
Yes I see your point and know that RFC4180 recommends to put the commas between quotes for proper CSV export. Does the CSV extension automatically add those commas while doing the export?

Here is for example a kind of value we can have. for "Lithology" : Calcaire gréseux très glauconieux, coquillé, plus ou moins friable, peu fracturé, beige   verdâtre à gris verdâtre.

Most of the time we change the separator for CSV export to solve the issue (generally using TAB instead). So my proposal was to enable to define the separator. I never faced a case where one would use , ; and TAB at the same time.

hylkevds, 2020-02-21:
Yes, barring bugs, the output will be RFC4180 compliant.
We probably should explicitly mandate that in the spec.

sgrellet, 2020-02-21:
+1, we should emphasize that in the spec. so far the RFC is just mentionned for how to handle properties and parameters fields

sgrellet, 2020-02-21:
+1 the implementation seems ok. your example already contains nasty combinations :) . we just tried the following

- https://ogctest.docker01.ilt-dmz.iosb.fraunhofer.de/v1.1/Things?$expand=Datastreams&resultFormat=CSV 
-> does not work according to spec
- https://ogctest.docker01.ilt-dmz.iosb.fraunhofer.de/v1.1/Things?$expand=Datastreams($top=1)&$resultFormat=CSV
-> works according to spec

hylkevds, 2020-02-25:
I've updated the spec to explicitly state that the resulting CSV must conform to RFC4180.