opengeospatial/sensorthings #57: Provide datetime arithmetic function for use in $filter expressions

selimnairb, 2018-11-16:
I may be misinterpreting some of OData's datetime capabilities, but if not, here is a use case I would like to see supported for $filter in STA 1.1.

As a dashboard or analytics developer, I would like to be able to query for Observations based on date/time arithmetic from `now()`.  For example, I would like to be able to query Observations within that last 5 minutes, or last 5 seconds, etc. without having to formulate and include the relevant datetime string my query.  So, if I want to fetch Observations that are less than 5 seconds old, instead of having to write this query:

```
https:/www.example.com/SensorThingsService/v1.0/Observations?$filter=phenomenonTime gt 2018-11-16T17:59:55Z
```
(assuming the current time is 2018-11-16T18:00:00Z)

I want to be able to say something like:

```
https:/www.example.com/SensorThingsService/v1.0/Observations?$filter=phenomenonTime gt dtincdec(now(), 'seconds', -5)
```
Where `dtincdec()` is short for increment or decrement a datetime.

Without this capability, there are some time queries that are difficult/impossible to do.  In the example above, if the request latency is greater than N seconds, I can't reliably ask the server for Observations that are at most N seconds old (at the time of query).

selimnairb, 2018-11-16:
Sorry @hylkevds here is another issue that has been kicking around in my head for the last few weeks ...

hylkevds, 2018-11-16:
You can already do that, using normal add and sub, with durations. I had a slide about that in my presentation on the summit.
Some examples from https://github.com/FraunhoferIOSB/FROST-Server/wiki/Example-queries :
## The previous day
The observations of a Datastream, for the last day:

```
Datastreams(1)/Observations?$filter=phenomenonTime gt now() sub duration'P1D'
```

## The last x days

The observations of a Datastream, for the last days, where the number of days is configured in properties/days of the Datastream:

```
Datastreams(1)/Observations?$filter=phenomenonTime gt now() sub duration'P1D' mul Datastream/properties/days
```

selimnairb, 2018-11-16:
@hylkevds As I suspected.  Thanks for the information.  I will close this.

closed 2018-11-16 by selimnairb