opengeospatial/sensorthings #65: Related Observations

KathiSchleidt, 2019-01-28:
Within O&M there is a relatedObservation concept; this is currently missing within STA. Based on our use case analysis, this type of relation is required not only between Datastreams (as the equivalent to the O&M Observation), but also between STA Observations and STA FoIs.
While this type of relationship information could be provided under the properties extension, defining an optional dedicated relatedObservation attribute on Datastream, Observation and FoI would be more explicit.

Link to Fraunhofer FROST GitHub: https://github.com/FraunhoferIOSB/FROST-Server/issues/96
Link to INSIDE-Information-Systems GitHub: https://github.com/INSIDE-information-systems/SensorThingsAPI/issues/5

hylkevds, 2024-01-25:
One of the ideas for V2 is a Relations Extension that does this:
[image]

hylkevds, 2025-03-28:
I've been thinking about the navigatonProperties pointing to the relations:
- `Thing/TargetRelatedThings`
- `Thing/SourceRelatedThings`

I think those names are not very clear. Which one points to a relation where our Thing is the source, and which points to the Relation where our thing is the Target? And these navigatonProperties don't actually point to Things...
An alternative option:

- `Thing/TargetOfRelations` (Relations with this Thing as Target)
- `Thing/SourceOfRelations` (Relations with this Thing as Source)

hylkevds, 2025-04-02:
### Current State
Features:
- Building: `Features(1)`
- Room: `Features(2)`

`RelationRole(1)`:
- isInBuilding (inverse: hasRooms)
  - Source: a Room
  - Target: a Building

`RelatedFeatures(1)` has 
- Source Room `Features(2)`
- Target Building `Features(1)`

From the Room: `Features(2)/TargetRelatedFeature/Target` would point to Building `Features(1) `
From the Building: `Features(1)/SourceRelatedFeature/Source` would point to Room `Features(2)`

### Proposal 1
Rename
- `RelatedFeature` to `FeatureRelation`
- TargetRelatedFeature to TargetFeatureRelation
- SourceRelatedFeature to SourceFeatureRelation

From the Room: `Features(2)?$expand=TargetFeatureRelations/Target` would point to Building `Features(1) `
From the Building: `Features(1)?$expand=SourceFeatureRelations/Source` would point to Room `Features(2)`