opengeospatial/sensorthings #66: Observation/parameters type

hylkevds, 2019-01-28:
This issue is just to document a change from v1.0 to v1.1, so it can be referenced.

The type of `Observation/parameters` is defined in STA v1.0 as

> NamedValues in a JSON Array

Since this would be hard to filter on (addressing array items) and to be more consistent with the rest of the API, the type of `Observation/parameters` has been changed to just "JSON Object" in v1.1.

This issue is already addressed by commit:
https://github.com/opengeospatial/sensorthings/commit/1218e535ec7a9e3aa8910b747d4eeac00b2d19f9

hylkevds, 2019-04-23:
Does this break any current implementations? How do FROST, SensorUp and Gost implement Observation/parameters ?

taniakhalafbeigi, 2019-04-23:
We are implementing it as JSON Object.
It was really a mistake to phrase it that way. But I am not sure if it affected other implementations.

hylkevds, 2019-04-24:
FROST also implements it as JSON Object.

bertt, 2019-04-24:
in GOST its also a JSON object

hylkevds, 2022-02-23:
Fixed with the publication of Sensing v1.1.

closed 2022-02-23 by hylkevds