opengeospatial/sensorthings #72: [DocumentationRequest} Machine Readable Version of Standard using OData CSDL Schema

riedel, 2019-04-05:
Many tools /particularly all the Microsoft .NET [LINQ Support](https://docs.microsoft.com/en-us/dotnet/framework/data/wcf/linq-considerations-wcf-data-services)) won't work on SensorThings Services without the related [CSDL Schema](http://docs.oasis-open.org/odata/odata-json-csdl/v4.0/odata-json-csdl-v4.0.html) available on $metadata. 

As this is a static file and SensorThings is OData compliant it should be possible to provide it as an annex to the standard.

riedel, 2019-04-05:
As a side note: from this one could probably also generate an OpenAPI interface using this official stylesheet: https://tools.oasis-open.org/version-control/browse/wsvn/odata/trunk/spec/examples/V4-CSDL-to-OpenAPI.xsl which would accelerate adoption

KathiSchleidt, 2019-04-06:
Hi Till,
as bridging the gap between the OData based STA and the soon to be OpenAPI based WFS3 is one of the currently burning issues, utilizing CSDL sounds especially interesting (what with "free OpenAPI conversion"!). You state that the CSDL (sorry, not familiar with this, too many different technologies buzzing around!) just requires a simple static file - as you seem to be familiar with CSDL, could you provide support in the creation of this file?

riedel, 2019-04-06:
It seems there was an actual effort on this reading the standard:

"Things API does not follow the OData Common Schema Definition Language and as a result does not follow its metadata service entity model. Thus, SensorThings API should not be seen as an OData compliant API.  SensorThings API’s future work will explore possible harmonization between SensorThings API and OData."

I am not familiar enough to understand what are the current show stoppers of making SensorThings OData compliant.

Maybe someone of the editors could comment on this and the possible needed changes in the standard.

liangsteve, 2019-04-08:
The reason was at the time STA about to be published, CSDL was introduced in a later version of OData.  As a result it wasn't included in STA and listed as a future work.

hylkevds, 2019-04-09:
I just had a quick look through the CSDL Schema definition, looking for those points that might cause issues.
* Arbitrary property objects are possible
* Observation/result of type `ANY` is possible (`"type":["number","string","array","object","boolean","null"]`)
* Geo types seem to be covered (docs.oasis-open.org/odata/odata-json-csdl/v4.0/csd01/schema/edm.json#/definitions/Edm.Geography)

The only possible problem that I see is:
* no time interval, but that could be solved by specifying a String with a validation regular expression

volcan01010, 2021-07-07:
I'd like to add a +1 for this.

We have sensor data served via FROST API.  I tried to use the [python-odata](https://github.com/SAP/python-pyodata) library to query it, but was unable to instantiate a Client class because it relies on the OData metadata file:

```python
>>> client = pyodata.Client(SERVICE_URL, requests.Session())
HttpError: Metadata request failed, status code: 404, body:
b'{"code":404,"type":"error","message":"Not a valid id: Path is not valid."}'
```

It was looking for the metadata file at: `FROST-Server/v1.1/$metadata`

hylkevds, 2021-07-07:
If you have experience with OData, could you create a `$metadata` document for the SensorThings API?
It should be a (mostly) static document, shouldn't it?

volcan01010, 2021-07-07:
Unfortunately, I don't have experience with OData aside from hand-written FROST Server requests.  Today was my first attempt at connecting via a dedicated library.

hylkevds, 2021-12-02:
As an experiment I've implemented OData 4.0 and 4.01 for a STA endpoint. Due to differences in the encoding, I've given these versions different URLs, to make it clear what is being used:

https://ogc-demo.k8s.ilt-dmz.iosb.fraunhofer.de/ODATA_4.0/
https://ogc-demo.k8s.ilt-dmz.iosb.fraunhofer.de/ODATA_4.01/

In theory, one could use the same URL for STA and different OData versions, since OData defines a header for clients to request a maximum OData version, but for these experiments I prefer making things explicit.

If you're still interested, can you try out these endpoints?
The 4.0 version seems to work fine in Excel 2019+ (For windows, Mac version have no OData support)

riedel, 2021-12-03:
Did you have to modify the actual API or is this just implementing the "/$metadata" URL?

hylkevds, 2021-12-03:
It does modify the API slightly, since there are small differences in the JSON keys for generated properties, as documented in #136.

volcan01010, 2021-12-03:
Thanks for doing this, I am interested to try it.  I had a quick attempt to connect, but it turns out that the "Enterprise-ready Python OData client" created by SAP only supports OData V2 and didn't like the JSON output.

I hadn't spotted that this library was so out-of-date before.  It seems to be the most established Python OData framework on GitHub, too.  I'll have a look at other clients later.

riedel, 2021-12-04:
Yes, @hylkevds, first of all many thx!! @volcan01010 : have you tried this PR SAP/python-pyodata#77 ?

riedel, 2021-12-05:
Actually I tried using both excel 2016 and linqpad 6 and I strangely get SSL validation errors on both. However, this error seems not uncommon: https://forum.linqpad.net/discussion/2413/linqpad-6-with-odata-4-error

hylkevds, 2021-12-06:
I'm pretty sure Excel 2016 only supports OData 3.0, not 4.0. OData 3.0 only has XML.
Certificate exceptions are always tricky without specific error, could be many things.

riedel, 2021-12-06:
The error, however, from[ meancrazy/LINQPadOData4](https://github.com/meancrazy/LINQPadOData4) which uses  Microsoft's [OData v4 Client Code Generator](https://visualstudiogallery.msdn.microsoft.com/9b786c0e-79d1-4a50-89a5-125e57475937).

hylkevds, 2021-12-06:
What headers does it send in its request to the landing page?

riedel, 2021-12-06:
I used 4.0. It pretty much sends no headers at all (except for the path). So I used mitm proxy to ask for XML, which seems to be the expected default also for the 4.0 client. There seems to be an easy way to validate metadata in .net btw (haven't tried): https://github.com/cmshawns/ODataValidator/blob/master/ODataValidator/ODataValidator.cs