opengeospatial/sensorthings #79: Update

liangsteve, 2019-07-15:
From the James at BGS:

`Hi Steve,

You should probably change GeoJSON mime type in Table 6 as this is now obsolete

Table 6 List of some code values used for identifying types for the encodingType of the
Location and FeatureOfInterest entity

GeoJSON application/vnd.geo+json

Ref:  https://tools.ietf.org/html/rfc7946

The media type for GeoJSON text is "application/geo+json" and is
   registered in the "Media Types" registry described in [RFC6838].  The
   entry for "application/vnd.geo+json" in the same registry should have
   its status changed to be "OBSOLETED" with a pointer to the media type
   "application/geo+json" and a reference added to this RFC.

James`

KathiSchleidt, 2019-07-17:
Allow any geo mime type, or if no mime type is available a URL should be provided (ideally referencing a namespace).
In Table 7, add "http://www.opengeospatial.org/standards/sfa" for WKT;

And as the server needs to understand the geometry encoding utilized, the server should be able provide a list of supported geo formats. While the server may be able to store and provide non-supported geo formats, spatial search functionality will probably not be possible

selimnairb, 2019-08-02:
@KathiSchleidt @hylkevds See https://github.com/opengeospatial/sensorthings/pull/85 for a partial fix.  I didn't go into allowing for any geo type as this has broader implications (e.g. for spatial search as @KathiSchleidt mentioned) and I didn't remember what we had decided to do there, especially in relation to #4.

hylkevds, 2019-08-20:
The "other geo mime types" option is probably something for either an Extension or 2.0.
Should be made a separate issue.

hylkevds, 2022-02-23:
Fixed with the publication of Sensing v1.1.

closed 2022-02-23 by hylkevds