opengeospatial/sensorthings #90: MQTT notify entity changed

mjacoby, 2019-11-18:
Section `14.2.2 Receive updates of a SensorThings entity with MQTT Subscribe` of the standard says

> MQTT Control Packet: Subscribe
> 
> **Topic Pattern:** RESOURCE_PATH_TO_AN_ENTITY
> 
> **Example Topic:** Datastreams(1)
> 
> **Response:** When a property of the subscribed entity is updated, the service returns a complete JSON representation of the updated entity.
> http://docs.opengeospatial.org/is/15-078r6/15-078r6.html#91

What exactly is considered to be a `property` here? Are those only the "primitive properties" or does this include navigation properties? E.g. if I subscribe to `/Datastreams(1)` and the Thing that is linked to that Datastream is changed, should this trigger a MQTT notification?

hylkevds, 2019-11-18:
Personally I would say that navigation properties are not included. If one is interested in the Thing of Datastream 1, the subscription should be on `v1.0/Datastreams(1)/Thing`.

The reason is that for `v1.0/Datastreams(1)` the updated entity is the Datastream, but the `complete JSON representation` of a Datastream does not give any meaningful information (not even the Id) about the Thing. So sending a Datastream representation to the user when the Thing of the Datastream changes does not tell the user anything.

taniakhalafbeigi, 2019-11-18:
I personally agree with Hylke as well.

closed 2020-01-21 by mjacoby