opengeospatial/sensorthings #91: Sampling Strategy

KathiSchleidt, 2019-11-19:
We have multiple requests for support of the sampling strategy within STA.
The current prototypes provide a relatedTo association in the properties extension of the FoI.
It has been determined as too complex to support in a good practice, as the server-side query functionality should be able to utilize this relation in requests.
Thus it could make sense to define sampling as an extension (allowing us to introduce this before we finalize STA 2.0)

mjacoby, 2019-11-19:
Just for clarification, by sampling strategy you refer to something like this https://w3c.github.io/sdw/proposals/ssn-extensions/#ultimate-feature-of-interest? Could you please elaborate on what you mean by sampling strategy?

KathiSchleidt, 2020-01-21:
requirement for providing additional links between FoIs and/or Things is widespread
Question is if all features have an auto-relation, relation to other feature types?
Are these links generic (only relatedTo), or are they semantic?
Do we need a simple sub-model for some of these links (related to samplingStrategy from O&M), or just the possibility to add

hylkevds, 2022-06-14:
Since Sampling will be part of the Observations, Measurements and Samples specification, it would be good to add it to STA.
To do that properly it would be really helpful if some domain experts that actually have experience with Sample could give their take on how to best add Sampling to STA. Ideally experts from several different domains, to make sure we can cover as many use cases as possible.

Who could help us with that?