opengeospatial/sensorthings #98: SensorThings API – oneM2M interworking

ingofriese, 2020-02-07:
Dear SensorThings API community:
In the H2020 project mySMARTLife we had to connect two data platforms based on different standards OGC/STA and oneM2M. Now, we would like to standardize the interworking. We try to open-up a regarded work item in oneM2M. It would be also great to gain support in terms of infos, input, comments and review from this community OGC sensorthings.
 
I’m not sure how the formal process within OGC/STA is or how to formalize it, Hylke was as friendly as to tell me that a new issue on your GIT would be a good start. Within oneM2M I tried to motivate the work item with the following text:   

SensorThings API (STA) is an Open Geospatial Consortium (OGC) standard providing an open and unified framework to interconnect sensing IoT devices, data, and applications over the Internet. Among others, STA is going to become an important standard in Smart City.
Usually cities have already experience with OGC standards because they have to chart areas and properties. So the usage of STA is a logic consequence because its also an OGC standard. STA is easy to implement, open source software is available. STA is going to be used in large deployments in several European cities (e.g. City of Hamburg).
The interworking with oneM2M broadens the reach of those platforms because OGC/STA-based platforms then can be connected with an oneM2M-based Sensor- or Actuator Systems. And a oneM2M CSE can also be used to interwork with STA-enabled Sensors. The interworking enables heterogeneous architectures within a city that are combining the capabilities and advantages of both standards. In IoT as well as in Smart City there won’t be somethings like “the one  standard” so connecting elements between different standards become more important for future Smart City Infrastructures.

rhzg, 2020-02-19:
I think that is a good idea. The interworking between STA and oneM2M has also been an issue in some of our projects at Fraunhofer IOSB. We are happy to support such a work item as far as we can.

liangsteve, 2020-06-18:
There is publication about this topic.  A prototype system was done for this research.  Here is the link to that paper (in Chinese, abstract in English) http://ir.lib.ncu.edu.tw:88/thesis/view_etd.asp?URN=106322088&fileName=GC106322088.pdf