opengeospatial/sensorthings #102: PhenomenonTime not updated in Multidatastream obs insertion

spezzi, 2020-06-09:
I'm using both **datastreams** and **multidatastreams** in my FROST installation and I've noticed that **phenomenonTime** is set only for datastreams. 
I've seen the 3 triggers fired in the CRUD operations on OBSERVATIONS table that run the procedures to update phenomenontime, resulttime and observed_area fields in DATASTREAMS table, but no functions for MULTIDATASTREAMS. 

Am I missing something? tnx

KathiSchleidt, 2020-06-09:
Well - it may be more efficient to put this in on the FROST GitHub as it's an implementation issue, nothing to do with the standard itself

https://github.com/FraunhoferIOSB/FROST-Server/issues

spezzi, 2020-06-09:
You're completely right! I messed up with too many open browser tabs! Sorry again

closed 2020-06-18 by liangsteve