opengeospatial/sensorthings #105: Error/Discrepancy in Sensing Entities Overview Image

SpeckiJ, 2020-08-27:
There is a error in the UML diagram/Overview Image with the name of the relation between HistoricalLocation and Thing.

- UML Diagram of the sensing Entities (Section 8.2) names the relation `things`. (see [here](https://github.com/opengeospatial/sensorthings/blob/18-088/src/18-088/images/SensingEntities.png))

- The Defintion of the HistoricalLocation (Section 8.2.3) models it as `thing` (see [here](https://github.com/opengeospatial/sensorthings/blob/18-088/src/18-088/08_entities.adoc))

This affects all past versions and current drafts including 18-088.

Note: Spotted while debugging requests where `/HistoricalLocations?$expand=Things` is used [here](https://github.com/linksmart/grafana-sensorthings-datasource/blob/00f1d5d0b94244e1c337470415b0914291c7bd35/src/datasource.js#L91)

hylkevds, 2020-08-27:
Indeed, good catch.
It should be `Thing`, since the cardinality is single. A HistoricalLocation has only one Thing.

hylkevds, 2021-01-20:
Fixed in the upcoming v1.1.

hylkevds, 2022-02-23:
Fixed with the publication of Sensing v1.1.

closed 2022-02-23 by hylkevds