opengeospatial/sensorthings #106: Confusion about `$select` and `$expand` interaction

SpeckiJ, 2020-08-28:
This Issue is based on a discussion in the STA Grafana Dashboard Plugin: https://github.com/linksmart/grafana-sensorthings-datasource/issues/36 Transferred here as it is a more general discussion. 

**Problem Statement:**
The STA apparently does not clearly define (as there is discussion about it) whether the fact that Section 9.3.1 states that `$select` binds stronger than `$expand` also means that all expanded Properties need to be explicitly listed in the `$select` statement, or are implicitly included. See the original Discussion in the link above for concrete examples.

As mentioned by @hylkevds in the original discussion the OData Spec notes that they are implicitly selected
> http://docs.oasis-open.org/odata/odata/v4.0/errata03/os/complete/part1-protocol/odata-v4.0-errata03-os-part1-protocol-complete.html#_Toc453752287
> 11.2.4.1 System Query Option $select
>The service returns the specified content, if available, along with any available expanded navigation properties, ...

While the Specification example shows them being selected explicitly
> [Example 2 in `$select` documentation](http://docs.opengeospatial.org/is/15-078r6/15-078r6.html#48)

I have checked the Implementations (those known to me) and they seem to be split on this.

Has implicit `$select`: [GOST](https://github.com/gost/server), [FROST](https://github.com/FraunhoferIOSB/FROST-Server) (I assume - correct me if I'm wrong @hylkevds)
Needs explicit `$select`: SensorUp (gathered from Demo Instance [here](http://toronto-bike-snapshot.sensorup.com/v1.0/Things?$expand=Locations&$select=name)), [52°North STA](https://github.com/52North/sensorweb-server-sta)

Disclaimer: I am professionally involved in the development of the [52°North STA](https://github.com/52North/sensorweb-server-sta).

hylkevds, 2020-08-28:
Thank you for opening the issue here! Also probably still in time for a fix in 1.1 :)

Since you mentioned you also think implicit selection of expanded navigation properties makes sense, I suggest we change the listed order of $select and $expand.
We may also want to update example 2 to use this implicit selection, and explicitly mention it.

Interestingly, this changed from OData V3 to V4. See:
V3: https://services.odata.org/V3/OData/OData.svc/Persons?$format=application/JSON&$expand=PersonDetail&$select=ID
V4: https://services.odata.org/V4/OData/OData.svc/Persons?$expand=PersonDetail&$select=ID

liangsteve, 2020-09-10:
In the STA SWG meeting on 2020-09-10, the SWG agrees to follow the OData v4 pattern. @hylkevds will draft the fix in v1.1 for the SWG to review.

hylkevds, 2020-09-11:
PR Made: https://github.com/opengeospatial/sensorthings/pull/107
Also added to the AllChanges PR.
Resulting document: [HTML](https://hylkevds.github.io/sensorthings/18-088-AllChanges/18-088.html) [PDF](https://hylkevds.github.io/sensorthings/18-088-AllChanges/18-088.pdf)

hylkevds, 2020-10-01:
Best to leave it open until the PR is actually merged, after the SWG has had a look at the details of the proposed changes.

hylkevds, 2022-02-23:
Fixed with the publication of Sensing v1.1.

commit 883120bb (2020-11-24): Fixes #106: Clarified that expanded items are automatically selected
closed 2022-02-23 by hylkevds