opengeospatial/sensorthings #109: Filter on empty geometry feature, empty object, or non-existing property

matcho, 2020-10-23:
Hello,

We have a set of `FeaturesOfInterest`, some of them are geolocated and others are not. In the latter case, the `feature` property is an empty object `{}` which satisfies the mandatory nature of the property.

Now we're trying to obtain the list of non-geolocated FeaturesOfInterest.

Is it possible to filter on "empty geometry feature", as does PostGIS `st_isempty()` function ?

Otherwise, is it possible to filter on "empty object", for ex. with something that could look like `$filter=feature eq {}` or `$filter=empty(feature)` ?

In an even more general way, is it possible to filter on "non-existing property" or "null property", for ex. `$filter=feature/type eq NULL` or `$filter=not has_property(feature/type)` ?

Or is there another way to achieve such queries with other semantics ?

Thank you for any advice,
Mathias

hylkevds, 2020-10-29:
An explicit check for null would be nice.
One thing that comes to mind that should already work is using the length() function to check the contents of the field:
```
    /v1.1/FeaturesOfInterest?$filter=length(feature) lt 5
```

hylkevds, 2020-10-29:
Also, since the feature field has type ANY, you can also use an empty string as feature stead of `{}`.
and you could set the encodingType to `"none"`, then you can filter on that.

matcho, 2020-11-05:
Thank you, filtering on `length(feature) lt 5` works well and fast.