opengeospatial/sensorthings #110: Filter by external geometry reference

KathiSchleidt, 2020-11-07:
It would be really valuable when specifying a geometry filter not to have to include the entire geometry within the request (pretty messy and ugly), but instead be able to reference a geoJSON feature with this geometry (similar to #31 on absolute references in filtering, but more widely usable) via URL.

Potential use case: all measurements within the flood-plain described by that feature over there.

This way one could utilize geometries from both other STA endpoints as well as feature from OGC API - Features (or anything else that resolves to geoJSON)

ksonda, 2022-06-06:
Just bumping this. Have an implementation that runs into the URL character limit when passing complex polygon geometries as filters

hylkevds, 2022-06-06:
@ksonda: can you try if using batch requets ([JSON](https://fraunhoferiosb.github.io/FROST-Server/extensions/JsonBatchRequest.html) or [default](http://docs.oasis-open.org/odata/odata/v4.0/os/part1-protocol/odata-v4.0-os-part1-protocol.html#_Toc372793748)) gets around the URL limit?

Allowing URLs in the geometry fields of queries may have security implications, since the server would be executing get requests on user-supplied URLs... It would require some robust filtering. (whitelisting? only relative URLs?)