opengeospatial/sensorthings #124: Cos4coud Citizen Science

hylkevds, 2021-04-26:
From the email of Joan:

From the Citizen Science community there is some interest in using STA. We also have detected some extra needs in the STA data model.

In particular, we would like to propose "party" as another class in the diagram (the party is the citizen). We also would like to be able to associat a licence to a datastream and to group observations into a "blog" that allows on providing "redundant" observations on the same fenomena (tipical thing in citizen science)

humaidkidwai, 2023-09-21:
Can this requirement not be accommodated with the current STA Model?

Thing/Host is the Party
Sensor/Observer is the singular person reporting the data
Datastream/ObservationCollection is the group of observations reported by the Party which comprises of all the citizens

hylkevds, 2023-09-22:
I guess the answer is in the STAPlus [best practice](https://docs.ogc.org/bp/21-068.pdf) and [OGC SensorThings API Extension: STAplus 1.0](https://docs.ogc.org/DRAFTS/22-022.html) that is about to be published :)
So I think this issue can be closed...