opengeospatial/sensorthings #134: List of public end-points

hylkevds, 2021-11-17:
Hi All,

In a recent discussion the question came up if we could make a list of public end-points. This would be very useful for giving demos and trying out the SensorThings API.

How would we organise such a list? Some relevant aspects:

- One big table / multiple tables
- Country / region information
- Domain information
- Publisher
- URL
- Writeable by all (sandbox services)
- Supported version(s)
- Used implementation

What do you all think?
Here are some quick examples to help the discussion (Markdown in issues doesn't support tables):

## Sandbox Services

| Owner | access | version | Endpoint URL
| Fraunhofer IOSB | CRUD | 1.1 | https://ogc-demo.k8s.ilt-dmz.iosb.fraunhofer.de/v1.1/
| SensorUp | CRUD | 1.0 | http://scratchpad.sensorup.com/OGCSensorThings/v1.0

## Germany
| Owner | Domain | Info | Endpoint URL
| Freie und Hansestadt Hamburg, Landesbetrieb Geoinformation und Vermessung | Smart Cities | [info](http://www.urbandataplatform.hamburg/) | https://iot.hamburg.de/v1.1/

## United Kingdom
| Owner | Domain | Info | Endpoint URL
| British Geological Survey | Ground water | [info](https://sensors.bgs.ac.uk/) | https://sensors.bgs.ac.uk/FROST-Server/v1.1

sgrellet, 2021-11-17:
Great idea, I'd do it by domain (as opposed to by country/region) as it may reasure new comers their domain is covered. 
And as, most of the time, you define a 'pattern' to match ST API to your domain terms/practices.
I tested this approach in Franch and it proved really helpful (see here : https://github.com/INSIDE-information-systems/SensorThingsAPI).

+1 on the rest of the item

I'd add a date info related to each implementation (available YYYYMMDD) to anticipate the fact that, over time, some endpoints may not be available anymore (ex : R&D projects)

ksonda, 2021-11-19:
Just to keep track, I have several in this viewer https://sta-client.internetofwater.dev

sgrellet, 2021-11-19:
@ksonda , great ! : you can add this one in your viewer -> French Surface Water Quality bank (snapshot from end of 2018) : https://sensorthings-wq.brgm-rec.fr/FROST-Server/v1.0 . That may be of interest for the internet of water activities.

hylkevds, 2021-11-23:
Per domain sounds good. Which domains would we want?

@ksonda: is there a way to see which endpoints are being used on that site? Since the endpoints are not accessed directly by the browser, but by your server.

## Sandbox Services

| Owner | access | version | Endpoint URL
| Fraunhofer IOSB | CRUD | 1.1 | https://ogc-demo.k8s.ilt-dmz.iosb.fraunhofer.de/v1.1/
| SensorUp | CRUD | 1.0 | http://scratchpad.sensorup.com/OGCSensorThings/v1.0

## Smart Cities
| County | Owner | Domain | Info | Endpoint URL
|Germany | Freie und Hansestadt Hamburg, Landesbetrieb Geoinformation und Vermessung | Smart Cities | [info](http://www.urbandataplatform.hamburg/) | https://iot.hamburg.de/v1.1/

## Water
| County | Owner | Domain | Info | Endpoint URL
| UK | British Geological Survey | Ground water | [info](https://sensors.bgs.ac.uk/) | https://sensors.bgs.ac.uk/FROST-Server/v1.1
| France | BRGM | Surface water quality | [info](https://www.sandre.eaufrance.fr/) | https://sensorthings-wq.brgm-rec.fr/FROST-Server/v1.0

ksonda, 2021-11-23:
They're here, but if you want to add a table in markdown document I can PR into it with the metadata you put above.

https://github.com/internetofwater/sta-dashboard/blob/main/endpoints.json