opengeospatial/sensorthings #137: Add JSON-LD Context to STA

KathiSchleidt, 2022-02-02:
As mentioned in the charter, investigation of JSON-LD within STA would be interesting. Here a first attempt, I created a context file covering all STA classes and attributes, popped it in the response from a STA providing Thing and ObsProp:
 - Context: https://gist.github.com/KathiSchleidt/e74b8eb7f8c8723bc3d81b13c220d525
 - Data linked to Context: https://gist.github.com/KathiSchleidt/6c897acd177109266e498fe7f42d2572

Here you can play with it in the JSON-LD playground:
https://json-ld.org/playground/#startTab=tab-expanded&json-ld=https%3A%2F%2Fgist.githubusercontent.com%2FKathiSchleidt%2F6c897acd177109266e498fe7f42d2572%2Fraw%2Fgistfile1.txt&context=%7B%7D

So - what have I done wrong? ;)

hylkevds, 2022-02-02:
Cool, nice work.

Does one Context document describe the entire Service, or does one need a different one for each "page"?

Both OData and JSON-LD define the "@context" property. We should check if that causes issues. Probably not, since OData clients clearly identify that they want OData and not LD.

KathiSchleidt, 2022-02-16:
Goal is to leave STA the way it is, find a way of expressing this in the context.
I admit that the example above is a hack, got tired of talking about it, as issues tend to show up in the doing, I just did (and the issues are nicely following :) )

First, a disclaimer: I created context entries for classes/attributes I could find a semi-sane correspondence (these still need to be checked!), then went freestyle where I couldn't find anything, reason for the dummy sta namespace linked to http://something.org/sta/ If anybody can find concepts for the bits I packed under this ns, please tell me!

@ksonda the geosparql bits in the context are leftover from the (S)ELFIE context I bastardized this from. During (S)ELFIE we were still dealing with JSON-LD V1.0, that had issues with encoding geometry (issue pertained to polygons. In contrast to JSON, LD doesn't include a sequence in arrays as it's modelled on RDF, means when you provided a polygon, you got an unordered set of coordinate pair, good luck finding the polygon ;) This has been resolved in LD V1.1 that allows for arrays to be either lists (ordered) or sets (grab-bag)). When I did the STA hack, I improvised badly:
- LOCATION: gsp:SpatialObject
- location & feature: schema:geo

In the STA implementations I've been using (FROST), the geometries in the location & feature attributes is provided as JSON Geometry, but as the STA data model specifies ANY for these types, you can implement as you please, e.g. geosparql WKT - define a context accordingly.

I'm still waiting for the JSON-FG spec to settle, see where OGC is going with JSON Geometries, then continue figuring. Happy for any and all insights!

KathiSchleidt, 2022-02-16:
@ksonda 2nd post on your reworked STA/OAF endpoint (COOL!) Questions:
- Any reason you're not doing a GeoJSON-LD encoding here? Would allow simple GeoJSON applications to get on with things while the more advanced applications can find their geometries anyway
- What's the reasoning for the 2 context objects? First one more general for the OAF, 2nd for STA?
- Links (still struggling with where O-APIs are going :( ): is the split between the @iot.selfLink linking back to the source STA while the general links stay within the OAF?
- How do you foresee querying this API? When I go back to the STA I can easily find that you're providing depth measurements, but with OAF I'd have to step through the model graph to get there

KathiSchleidt, 2022-06-16:
Collect insights at https://docs.google.com/document/d/1EJzo10eO1m6yQFvQfrB5SZuxTOL-ZFd39O7VKOfKsrk/edit#heading=h.3jf18irghtme

ksonda, 2023-10-03:
Do you have an example document of how this would work? I don't think parsers could directly use it that way.

ksonda, 2023-10-03:
Let me clarify. I think no matter what "@context" has to be at the "top" level of the hierarchy of JSON document. You can accomplish something like the properties block solution you describe via [context nesting](https://www.w3.org/TR/json-ld11/#nested-properties), for one level of hierarchy only. So, something like this could work:

Consider the STA Thing:
https://labs.waterdata.usgs.gov/sta/v1.1/Things('AR008-331856091114601')

You can add this context up top:

```
"@context": {
    "schema": "http://schema.org/",
    "name": "schema:name",
    "@iot.selfLink":"@id",
    "properties": "@nest",
     "monitoringLocationUrl": {
       "@id": "schema:url",
        "@nest": "properties"
        }
  }
```

[JSON-LD playground](https://json-ld.org/playground/#startTab=tab-compacted&json-ld=%7B%22%40context%22%3A%7B%22schema%22%3A%22http%3A%2F%2Fschema.org%2F%22%2C%22name%22%3A%22schema%3Aname%22%2C%22%40iot.selfLink%22%3A%22%40id%22%2C%22properties%22%3A%22%40nest%22%2C%22monitoringLocationUrl%22%3A%7B%22%40id%22%3A%22schema%3Aurl%22%2C%22%40nest%22%3A%22properties%22%7D%7D%2C%22%40iot.selfLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%22%2C%22%40iot.id%22%3A%22AR008-331856091114601%22%2C%22name%22%3A%22AR008-331856091114601%22%2C%22description%22%3A%22Well%22%2C%22properties%22%3A%7B%22state%22%3A%22Arkansas%22%2C%22active%22%3Atrue%2C%22agency%22%3A%22Arkansas%20Natural%20Resources%20Commission%22%2C%22county%22%3A%22Chicot%20County%22%2C%22country%22%3A%22US%22%2C%22district%22%3A%22Arkansas%22%2C%22stateFIPS%22%3A%22US%3A05%22%2C%22wellDepth%22%3A%2280%22%2C%22agencyCode%22%3A%22AR008%22%2C%22countyFIPS%22%3A%22US%3A05%3A017%22%2C%22countryFIPS%22%3A%22US%22%2C%22districtCode%22%3A%2205%22%2C%22altitudeDatum%22%3A%22North%20American%20Vertical%20Datum%20of%201988%22%2C%22altitudeMethod%22%3A%22Interpolated%20from%20Digital%20Elevation%20Model%22%2C%22hydrologicUnit%22%3A%22080500020302%22%2C%22altitudeAccuracy%22%3A%221.6%22%2C%22monitoringLocationUrl%22%3A%22https%3A%2F%2Fwaterdata.usgs.gov%2Fmonitoring-location%2F331856091114601%22%2C%22monitoringLocationName%22%3A%2216S01W10CC1%20CH-32%20WU%22%2C%22monitoringLocationType%22%3A%22Well%22%2C%22monitoringLocationNumber%22%3A%22331856091114601%22%2C%22monitoringLocationAltitudeLandSurface%22%3A%22124%22%7D%2C%22Locations%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FLocations%22%2C%22HistoricalLocations%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FHistoricalLocations%22%2C%22Datastreams%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FDatastreams%22%7D&frame=%7B%7D&context=%7B%7D)

However, as soon as you do anything where STA returns the response nested in a "value" array, the same context would not work.

For example. if the query was https://labs.waterdata.usgs.gov/sta/v1.1/Things?$top=1 , see the result:

[Example JSON-LD playground](https://json-ld.org/playground/#startTab=tab-compacted&json-ld=%7B%22%40context%22%3A%7B%22schema%22%3A%22http%3A%2F%2Fschema.org%2F%22%2C%22name%22%3A%22schema%3Aname%22%2C%22%40iot.selfLink%22%3A%22%40id%22%2C%22properties%22%3A%22%40nest%22%2C%22monitoringLocationUrl%22%3A%7B%22%40id%22%3A%22schema%3Aurl%22%2C%22%40nest%22%3A%22properties%22%7D%7D%2C%22value%22%3A%5B%7B%22%40iot.selfLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%22%2C%22%40iot.id%22%3A%22AR008-331856091114601%22%2C%22name%22%3A%22AR008-331856091114601%22%2C%22description%22%3A%22Well%22%2C%22properties%22%3A%7B%22state%22%3A%22Arkansas%22%2C%22active%22%3Atrue%2C%22agency%22%3A%22Arkansas%20Natural%20Resources%20Commission%22%2C%22county%22%3A%22Chicot%20County%22%2C%22country%22%3A%22US%22%2C%22district%22%3A%22Arkansas%22%2C%22stateFIPS%22%3A%22US%3A05%22%2C%22wellDepth%22%3A%2280%22%2C%22agencyCode%22%3A%22AR008%22%2C%22countyFIPS%22%3A%22US%3A05%3A017%22%2C%22countryFIPS%22%3A%22US%22%2C%22districtCode%22%3A%2205%22%2C%22altitudeDatum%22%3A%22North%20American%20Vertical%20Datum%20of%201988%22%2C%22altitudeMethod%22%3A%22Interpolated%20from%20Digital%20Elevation%20Model%22%2C%22hydrologicUnit%22%3A%22080500020302%22%2C%22altitudeAccuracy%22%3A%221.6%22%2C%22monitoringLocationUrl%22%3A%22https%3A%2F%2Fwaterdata.usgs.gov%2Fmonitoring-location%2F331856091114601%22%2C%22monitoringLocationName%22%3A%2216S01W10CC1%20CH-32%20WU%22%2C%22monitoringLocationType%22%3A%22Well%22%2C%22monitoringLocationNumber%22%3A%22331856091114601%22%2C%22monitoringLocationAltitudeLandSurface%22%3A%22124%22%7D%2C%22Locations%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FLocations%22%2C%22HistoricalLocations%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FHistoricalLocations%22%2C%22Datastreams%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FDatastreams%22%7D%5D%2C%22%40iot.nextLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings%3F%24top%3D1%26%24skip%3D1%26%24orderby%3D%2540iot.id%2Basc%26%24skipFilter%3D%2528%2540iot.id%2Bgt%2B%2527AR008-331856091114601%2527%2529%22%7D&context=%7B%7D)

ksonda, 2023-10-03:
However, supposing you could set a default vocabulary that covers all the attributes/properties you want to use (and have definition server describing all these properties), you could do something like this:

"@context": {
    "@vocab": "https://my-sta.org/vocab/",
    "schema": "http://schema.org/",
    "name": "schema:name",
    "@iot.selfLink":"@id",
    "properties": "@nest",
     "monitoringLocationUrl": {
       "@id": "schema:url",
        "@nest": "properties"
        }
  }

[Example JSON-LD Playground](https://json-ld.org/playground/#startTab=tab-compacted&json-ld=%7B%22%40context%22%3A%7B%22%40vocab%22%3A%22https%3A%2F%2Fmy-sta.org%2Fvocab%2F%22%2C%22schema%22%3A%22http%3A%2F%2Fschema.org%2F%22%2C%22name%22%3A%22schema%3Aname%22%2C%22%40iot.selfLink%22%3A%22%40id%22%2C%22properties%22%3A%22%40nest%22%2C%22monitoringLocationUrl%22%3A%7B%22%40id%22%3A%22schema%3Aurl%22%2C%22%40nest%22%3A%22properties%22%7D%7D%2C%22%40iot.selfLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%22%2C%22%40iot.id%22%3A%22AR008-331856091114601%22%2C%22name%22%3A%22AR008-331856091114601%22%2C%22description%22%3A%22Well%22%2C%22properties%22%3A%7B%22state%22%3A%22Arkansas%22%2C%22active%22%3Atrue%2C%22agency%22%3A%22Arkansas%20Natural%20Resources%20Commission%22%2C%22county%22%3A%22Chicot%20County%22%2C%22country%22%3A%22US%22%2C%22district%22%3A%22Arkansas%22%2C%22stateFIPS%22%3A%22US%3A05%22%2C%22wellDepth%22%3A%2280%22%2C%22agencyCode%22%3A%22AR008%22%2C%22countyFIPS%22%3A%22US%3A05%3A017%22%2C%22countryFIPS%22%3A%22US%22%2C%22districtCode%22%3A%2205%22%2C%22altitudeDatum%22%3A%22North%20American%20Vertical%20Datum%20of%201988%22%2C%22altitudeMethod%22%3A%22Interpolated%20from%20Digital%20Elevation%20Model%22%2C%22hydrologicUnit%22%3A%22080500020302%22%2C%22altitudeAccuracy%22%3A%221.6%22%2C%22monitoringLocationUrl%22%3A%22https%3A%2F%2Fwaterdata.usgs.gov%2Fmonitoring-location%2F331856091114601%22%2C%22monitoringLocationName%22%3A%2216S01W10CC1%20CH-32%20WU%22%2C%22monitoringLocationType%22%3A%22Well%22%2C%22monitoringLocationNumber%22%3A%22331856091114601%22%2C%22monitoringLocationAltitudeLandSurface%22%3A%22124%22%7D%2C%22Locations%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FLocations%22%2C%22HistoricalLocations%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FHistoricalLocations%22%2C%22Datastreams%40iot.navigationLink%22%3A%22https%3A%2F%2Flabs.waterdata.usgs.gov%2Fsta%2Fv1.1%2FThings('AR008-331856091114601')%2FDatastreams%22%7D&context=%7B%7D)

Wild idea is operationally for implementations, not necessarily part of the STA standard: Say FROST has a configurable /vocab endpoint that would serve as a simple definition server. e.g. /vocab/foo1 points to a JSON-LD document with a definition of /foo1, and if applicable, that it is owl:sameAs whatever schema.org or OGC definition server URI.

hylkevds, 2023-10-03:
> Let me clarify. I think no matter what "@context" has to be at the "top" level of the hierarchy of JSON document.

Reading the spec [4.1 Advanced Context Usage](https://www.w3.org/TR/json-ld/#advanced-context-usage):
> In general, contexts may be used any time a [map](https://infra.spec.whatwg.org/#ordered-map) is defined.

I interpret that as that an `@context` does not have to appear at the top of the document. The example also shows multiple Objects with different contexts. So each properties Object can have its own `@context`.

ksonda, 2023-10-04:
It's true you can put contexts anywhere, but they will only be interpreted if there is context declaring what a URI for the key at least one level above the JSON hierarchy where the desired context is. So you would have to declare that "properties" is something in a context at a level at least one above in the JSON hierarchy for piped-in context at the properties level to work. And given that many responses are actually in a "value" object, I think you'd need at minimum to have STA give a minimal context like this at the same level of any response:

```json
{
  "@context": {
    "value":"https://sta.org/response",
    "properties": "https://sta.org/properties"
  }
}
```

For example, consider https://labs.waterdata.usgs.gov/sta/v1.1/Things?$top=2&$select=@iot.id,name,properties/district, where we could pipe in context declaring `properties/district` to be something

```json
{
  "value": [
    {
      "@iot.id": "AR008-331856091114601",
      "name": "AR008-331856091114601",
      "properties": {
        "@context":{
            "district": {
                "@id":"https://schema.org/areaServed", 
                "@type":"https://schema.org/AdministrativeArea"
            }
         },
        "district": "Arkansas"
      }
    },
    {
      "@iot.id": "AR008-334933091153501",
      "name": "AR008-334933091153501",
      "properties": {
          "@context":{
            "district": {
                "@id":"https://schema.org/areaServed", 
                "@type":"https://schema.org/AdministrativeArea"
            }
         },
        "district": "Arkansas"
      }
    }
  ],
  "@iot.nextLink": "https://labs.waterdata.usgs.gov/sta/v1.1/Things?$top=2&$skip=2&$select=%40iot.id,name,properties%2Fdistrict&$orderby=%40iot.id+asc&$skipFilter=%28%40iot.id+gt+%27AR008-334933091153501%27%29"
}
```

This would not be interpreted. We would need to do this to be interpreted

```json
{
  "@context": {
    "value": "https://sta.org/response",
    "properties": "https://sta.org/properties"
  },
  "value": [
    {
      "@iot.id": "AR008-331856091114601",
      "name": "AR008-331856091114601",
      "properties": {
        "@context": {
          "district": {
            "@id": "https://schema.org/areaServed",
            "@type": "https://schema.org/AdministrativeArea"
          }
        },
        "district": "Arkansas"
      }
    },
    {
      "@iot.id": "AR008-334933091153501",
      "name": "AR008-334933091153501",
      "properties": {
        "@context": {
          "district": {
            "@id": "https://schema.org/areaServed",
            "@type": "https://schema.org/AdministrativeArea"
          }
        },
        "district": "Arkansas"
      }
    }
  ],
  "@iot.nextLink": "https://labs.waterdata.usgs.gov/sta/v1.1/Things?$top=2&$skip=2&$select=%40iot.id,name,properties%2Fdistrict&$orderby=%40iot.id+asc&$skipFilter=%28%40iot.id+gt+%27AR008-334933091153501%27%29"
}
```

But this IMHO is (a) involves a lot of redundant information being presented and (b) means we may as well put everything in a top-level context anyway.

The other issue is that this would be interpreted like this:

[image]

I'm not sure if this really serves any particular use case. If the use case is just to describe all the properties, then this would provide that information to a user, but not due to anything in particular about JSON-LD or RDF. May as well just pipe in an arbitrary JSON object called "data dictionary". 

If the use case is to provide a knowledge graph about each `Thing` in this case, then a graph node identifier for the `Thing` should be provided within the `properties` block.

KathiSchleidt, 2026-01-07:
Nice!
Question - Is there also an ambition to rework the STA contexts as discussed, shifting the core STA parts to a general STA context, then using the class specific contexts for properties extensions? At present, this is still mixed, muddling implementation specific bits with more general STA context.

Also noticed some glitches with how the STA running at Poznan is integrated. The links in the examples (e.g. [Datastream](https://british-oceanographic-data-centre.github.io/bblocks-sta/bblock/ogc.api.sta.Datastream/examples)) don't resolve, you're missing the single quotes around the string guid ID.

`https://sensor-things-api-sensor-things-api.apps.dcw1.paas.psnc.pl/jf2024/api/v1.0/Datastreams(2e92962c0b6996add9517e4242ea9bdcd2511e0ff7ac61981c6051a52b51f05c)/Thing`

doesn't resolve

When I add quotes (and an ID the system recognizes), it works:
`https://sensor-things-api-sensor-things-api.apps.dcw1.paas.psnc.pl/jf2024/api/v1.1/Datastreams('00c17237d011cca999f55a43db2ce040*d2511e0ff7ac61981c6051a52b51f05c')/Thing`

I'm aware of the glitch in linking Thing to Datastream, have some approaches, happy to discuss.