opengeospatial/sensorthings #147: STAplus: Should License be on Datastream or Observation or both?

securedimensions, 2022-07-05:
The current data model for the `plus` extension is the following (`MultiDatastream` not shown for simplicity): 

[image]

The `License` class is associated with `Datastream`. But, would it be "better" to associate the `License` class with `Observation`?

hylkevds, 2022-07-06:
Discussion at the 2022-07-06 Telco:

- Issue with License attached to Datastream:
  - Proliferation of Datastreams, since every observation a user makes can have a different license.
- Issue with License attached to Observation:
  - Inefficient data storage
  - No easy sharing of all Observations in a Datastream, since they may differ in License
  - In most use cases all Observations in a Datastream will have the same License any way
- Andreas has a data set with user generated data with many licenses. Analysis of this data set would be helpful.
- The License applies to the Observations in a Datastream, not to the other entities attached to the Datastream.

Related discussion for Party: Does each Observation need a different Party, or should it be attached to Datastream, or even only to Thing?
- Datastream can have a different Party than Thing: A satellite can have Sensors from different owners.
- How may cases are there where a Datastream has Observations from different Parties?
  - So far not. It is probably best to use Group for that.

Datastream may need to be attached to multiple Projects
- There may be an issue when those Projects have different runtimes.
- Project may need to be renamed to Campain?
- In some cases you do want to have Observations attached to different Campains
- In some cases it is much more efficient to have the Datastream be attached to a Project/Campain
- Adding some additional properties (termsOfUse, privacyStatement) to Group makes it possible to use a Group to represent a Campain

securedimensions, 2023-05-04:
After having modified the STAplus draft standard quite a bit since this issue was created, the following conclusions can be made:

> Issue License attached to Datastream and Observations

In STAplus, there are two different options to create a bag or set of observations:
* Per Datastream: For most use cases, it seems to be sufficient that all observations of the same sensor have the same license. So it is sufficient to attach the license to the Datastream.
* Per Group: For use cases that require to create a bag or set of observations independent from a Datastream (sensor), a Group can be used. Such a group can include observations with different licenses (as per the attached datastream). See example camera trap in the STAplus Best Practices that illustrates the use of observation groups in more detail. 

> Related discussion for Party: Does each Observation need a different Party, or should it be attached to Datastream, or even only to Thing?

It is important that a Thing and a Sensor can be operated by different users. However, all observations generated by the sensor belong to the same Party attached to the corresponding datastream. 

> Datastream may need to be attached to multiple Projects

According to the current data model, it is possible that the same datastream instance is attached to multiple projects. Therefore, the resulting observations can be associated to different projects

> Project may need to be renamed to Campaign?

Renaming the Project to Campaign would put the wording too much into Citizen Science. However, the use of STAplus is not only limited to Citizen Science. Therefore, the name Project should not be changed.

[image]
STAplus Data Model for `Datastream`

[image]
STAplus Data Model for `MultiDatastream`

closed 2023-05-04 by securedimensions