opengeospatial/sensorthings #152: STAplus and personal data

securedimensions, 2022-11-05:
Items for discussion:

* Can the `personalData` be used in `$filter` expressions?

* How can a party allow that their `personalData` can be used in `$filter`?

* An implementation shall not allow to use `$select` based on `personalData`!?

* An implementaiton shall not allow `$expand` on `personalData`!

* An authenticated user SHALL be able to access it's own `personalData` - that needs to become a REQUIREMENT in Authentication!

joanma747, 2023-05-04:
Lets add a "security considerations" section saying that in absence of any business logic about parties, personalData should not be available in $filter ... (because I could use repetitive filters to "reveil" personal data). A business logic could allow for a Party to filter its own PArty data but not the others.

Filter by authID is allowed because it is only name that does not show anything about the person.

securedimensions, 2023-05-04:
The standard is updated in that respect now.

closed 2023-05-04 by securedimensions