opengeospatial/sensorthings #156: STAplus License

securedimensions, 2023-03-30:
# STAplus needs to support custom attribution text

## Use Case
STAplus supports the association of a license with the entities `Datastream`, `Group`, and `Project`. For attribution licenses, the re-use of the data requires to craft an attribution text. When using Creative Commons licenses, a [recommended best practices](https://wiki.creativecommons.org/wiki/Recommended_practices_for_attribution) can be followed. 

However, in rare cases the creator of the `Datastream`, `Group`, or `Project` may want a custom attribution text to be used. In this case, the attribution text must be stored somewhere.

## Proposed Solution
To support this use case, it is RECOMMENDED to store the custom attribution text as a property named `attributionText` in the `properties` of the entity `Datastream`, `Group`, or `Project`.

securedimensions, 2023-05-04:
# Alternative Solution

An alternative solution could recommend that the `attributionText` gets stored within the `Properties` object for the `License` entity. But such a solution might lead to license proliferation if each user creates the own license. In order to ensure that only compatible licenses get created by users, the Business Logic could define proper constraints. Let's illustrate this in an example based on Creative Commons licenses:

* The implementation creates the seven recommended licenses from Creative commons.
* Each license instance is identified by a unique `definition` URI. 
* The implementation can control the creation of a license by checking the `definition` URI to make sure the license falls into the allowed licenses and only limit the created `License` to only contain the `attributionText`.
* This prevents the proliferation of licenses types (all licenses belong to the Creative Commons framework) and all licenses' compatibility can be determined via the `definition` URI.
* But the re-use of objects requires to use the `attributionText`.

# Data Model Updating Proposal
The ensure interoperability for the above implementation option, the `License` entity should include the property `attributeText` and the STAplus standard should recommend the solution above.

In addition, the `License` entity property `properties` (JSON Object) should be removed to prevent proliferation of incompatible licenses!

joanma747, 2023-05-04:
There is a precedent of removing properties from "Party" to pretect personal information. WE will have the same here to prevent the proliferation of "modifications" of standard licencies by adding properties.

We need attribution in here and I'm fine to use attributionText in licence. My first impression was that attribution belong to "Party" but I'm fine to have it in license.

securedimensions, 2023-05-04:
The `attributionText` property is now added to the `License`. And the `properties` property is removed.

closed 2023-05-04 by securedimensions