opengeospatial/sensorthings #160: Sensor vs. OMS:ObservingProcedure

KathiSchleidt, 2023-06-27:
If we add OMS:ObservingProcedure to the STA 2.0 model, we may want to make the cardinalities on both Sensor and OMS:ObservingProcedure, make them conditional on each other (one of the 2 must be provided)

liangsteve, 2023-09-27:
Or we can only keep one.  At the time we use "Sensor" because in the IoT community, no one know the "Procedure" concept.  Using "Sensor" was a pragmatic approach.  However, I think SensorThings API has established its own ecosystem and also the state of "IoT Platforms" changed.  IMHO, we can align more closely with OMS now.  My 2 cents.

hylkevds, 2024-01-25:
The current proposal has an ObservingProcedure (and Deployment) EntityType. Their use is optional.
It's currently missing the link ObservingProcedure <-> ObservedProperty relation.