opengeospatial/sensorthings #166: Testing compliance with OGC API and OAS

rob-metalinkage, 2023-09-27:
In general a statement regarding compatibility or conformance with OAS or other OGC API components should be supported by an automated CI/CT process.

OGC COSI team is developing a framework for reusable [OGC BuildingBlocks](https://opengeospatial.github.io/bblocks) and will be testing in multiple scenarios over the coming year.

Another  starting point will be a template for new building blocks - see [](https://github.com/opengeospatial/bblock-template/blob/master/USAGE.md)

we will also be developing reports (and will integrate tooling) for compliance testing approaches for OAS.

rob-metalinkage, 2023-09-28:
OpenAPI Specification, the underlying technology of OGC APIS .. (maybe you know it as Swagger?) 

https://swagger.io/specification/

cportele, 2023-09-28:
@rob-metalinkage 

You have referenced copies of the spec published by SmartBear. The official spec of OpenAPI v3.0 is at https://spec.openapis.org/oas/v3.0.3.

It is v3.0 that uses a fork of JSON Schema draft 5 with changes/extensions like `nullable`. v3.1 is a breaking change and uses the latest JSON Schema 2020-12.

hylkevds, 2023-09-28:
I always found OpenAPI to be short enough to not need acronymisation. Kathi thought OAS was OGC Abstract Specification...
OGC needs an official acronym list!

If someone wants to play around with an OpenAPI specification for STA, FROST can generate one. For example: https://airquality-frost.k8s.ilt-dmz.iosb.fraunhofer.de/v1.1/api
You can also get the POST/PUT/PATCH bits with:
https://airquality-frost.k8s.ilt-dmz.iosb.fraunhofer.de/v1.1/api?editing=true