opengeospatial/sensorthings #167: STA V2.0 Design Draft

humaidkidwai, 2023-09-28:
I designed a draft data model that addresses some of the concerns the community has had over the past couple of years and also attempts to align with OMS V3. 

[image]

**ObservationCollection**
In TC Singapore, perhaps this was debated as to whether or not is it really a collection of observations or just a Datastream renamed to ObservationCollection. 
[1] As per OMS, an ObservationCollection should be homogeneous or summarizing type. 
[2] Should ObservationCollection be allowed to contain different types of observations that may potentially violate the OBSC req set out by OMS? Will it help in addressing issues?
[3] resultType is adopted as per SWE Common JSON schema (see #122 )

**Deployment**
This entity describes the current state of observers (read Sensors) in their association with Host (Thing (?)) 
[1] position property may be used to describe geographic location of sensors for moving things?

**ObservingProcedure**
See #160 
Do we need it?

**FeatureOfInterest**
In order to introduce sampling in STA, FoI can be specialized further to pFOI and uFOI as specified in OMS. This also allows to extend pFOI by introducing and linking Sampling entitites with it. pFOI may act as a Sample proxy in the design

Please feel free to critique and put forward your suggestions

hylkevds, 2024-01-24:
Current full work-in-progress proposal:
[image]

hylkevds, 2024-02-20:
Updated image:

[image]

humaidkidwai, 2024-02-20:
resultTime::Observation needs to be of type TM_Instant. As described in #177 the way OData defines it is DateTimeOffset, this should be elaborated in the OData binding section of the standard and not in the data model.

hylkevds, 2024-02-21:
I fixed the image.

humaidkidwai, 2024-03-04:
Does the '+' sign indicate that the given property is mandatory? If yes, are we sure about making observedArea::Datastream, phenomenonTime::Datastream, etc. as optional properties?

humaidkidwai, 2024-03-06:
Is the Deployment entity to be included in the Core Sensing Part of the document?

hylkevds, 2024-03-14:
Updated images:
  observationType --> resultType
  deploymentTime --> time
  samplingTime --> time
  samplingLocation --> location
  Sampling/parameters --> properties
  Sampling/location type --> ANY

hylkevds, 2024-04-10:
Changed Sampling -> Thing relation cardinality to 0..1. 
There is no need to require a Thing/Host for a Sampling, since the domain relation SampledFeature is mandatory.

hylkevds, 2024-06-26:
I've created a prototype for testing with this data model:

https://ogc-demo.k8s.ilt-dmz.iosb.fraunhofer.de/FROST-MODELV2/

It has the V2 proposed data model with the v1.1 API.

michcfr, 2024-07-18:
> I've created a prototype for testing with this data model:
> 
> https://ogc-demo.k8s.ilt-dmz.iosb.fraunhofer.de/FROST-MODELV2/
> 
> It has the V2 proposed data model with the v1.1 API.

Hello, would it be possible to download the FROST-MODELV2 prototype from github?

hylkevds, 2024-07-18:
> Hello, would it be possible to download the FROST-MODELV2 prototype from github?

Yes, you can use the Docker images for it: 
- http: [frost-server-http:develop-dataModelV2-2.4.0-SNAPSHOT](https://hub.docker.com/layers/fraunhoferiosb/frost-server-http/develop-dataModelV2-2.4.0-SNAPSHOT/images/sha256-5179b85d5f536dbc1af1539e0715cc4db9eb77bb0d03e019952e9be8c01846a7?context=explore) 
- mqtt: [frost-server-mqtt:develop-dataModelV2-2.4.0-SNAPSHOT](https://hub.docker.com/layers/fraunhoferiosb/frost-server-mqtt/develop-dataModelV2-2.4.0-SNAPSHOT/images/sha256-d146896fa745c71cd708979257acbea8c06fc32a71d371e03b2f8c958793aa4e?context=explore)
- All in one [frost-server:develop-dataModelV2-2.4.0-SNAPSHOT](https://hub.docker.com/layers/fraunhoferiosb/frost-server/develop-2.x-2.4.0-SNAPSHOT/images/sha256-19fe95046c54f97b52d38cb0d0e629cf365103e399eb77b9f9ab5ab9f336475a?context=explore)

Or build it yourself from the [develop-dataModelV2](https://github.com/FraunhoferIOSB/FROST-Server/tree/develop-dataModelV2) branch

Enable the V2 model with the environment variables:
```
# Disable V1 model
plugins_coreModel_enable=false
# Enable model loader, needed for the v2 plugin
plugins_modelLoader_enable=true
# Enable V2 model
plugins_coreModelV2_enable=true
# Enable the (optional) OM model
plugins_modelOM_enable=true
# Enable the (optional) Sampling model
plugins_modelSampling_enable=true
# Enable the (optional) Relations model
plugins_modelRelations_enable=true
```

The database is currently not compatible with the V1 model, so don't put this over an existing V1.1 installation. Things may also break in the future...

KathiSchleidt, 2024-07-22:
Shouldn't we have the core trinity (`name`, `definition`, `description`) of STA attributes on ALL Classes.

Somehow `PreparationStep `got `link `in place of `definition`, `Sensor` has never had a `definition`. Anything against doing this?

hylkevds, 2024-07-22:
I've put the build of my latest draft at: https://hylkevds.github.io/23-019/23-019.html

hylkevds, 2024-07-25:
> Shouldn't we have the core trinity (`name`, `definition`, `description`) of STA attributes on ALL Classes.
> 
> Somehow `PreparationStep `got `link `in place of `definition`, `Sensor` has never had a `definition`. Anything against doing this?

I think adding a place for an external identifier is a good idea. That also makes it easier to link to other APIs.
For Core this would be all classes except Observation and HistoricalLocation? Or should Observation also get one?

humaidkidwai, 2024-10-28:
> For Core this would be all classes except Observation and HistoricalLocation? Or should Observation also get one?

I am of the view that Observation should be lean and lightweight for the sensors to push to the server. HistoricalLocation _may_ benefit from `definition` and _shall_  be more helpful with `description`

hylkevds, 2024-10-29:
definition is a link to an external reference. metadata is also meant to contain the actual SensorML description. There is overlap, since the definition could point to a sensorML description too, but in cases where people do not have an extra external definition server, having the data in the entity would be valuable.

KathiSchleidt, 2025-02-26:
Link from DS to Deployment is multiple, should be `deployments`

humaidkidwai, 2025-03-05:
> definition is a link to an external reference. metadata is also meant to contain the actual SensorML description. There is overlap, since the definition could point to a sensorML description too, [...]

In that case, shouldn't `metadata` be an optional attribute? Currently, it is mandatory so we are forcing the users to define the description using the `metadata` attribute even if it is a pdf file that describes the Sensor.

hylkevds, 2025-03-05:
Yeah, good point. Does anyone have objections to making metadata (and thus also encodingType) optional?

hylkevds, 2025-03-06:
If the metadata attribute is empty, the encodingType is pointless, so making encodingType mandatory when metadata is not, is counter-intuitive.

humaidkidwai, 2025-03-06:
My perspective was about using the `encodingType` attribute for identifying the type of metadata defined in the URI from the `definition` attribute, since a URI may point to any PDF/SensorML/HTML/Plain Text document.

hylkevds, 2025-03-07:
The encodingType is for the metadata attribute, not for the definition attribute. The definition itself is an identifier, and doesn't even need to resolve. That's why it is a URI and not a URL.