opengeospatial/sensorthings #174: Data Quality model extension

hylkevds, 2024-01-25:
Since no one really knows what to do with the resultQuality DQ_Element property on Observation, we need something better.
Some proposals are currently being worked on.

securedimensions, 2024-02-01:
One proposal / idea results from the work in the EU CitiObs project. It is understood that the use of the DQ_Element or the use of the Observation `parameters` puts heavy burdon on the database when filtering on Observations WHERE data-quality equals xyz. 

One possible way to improve filtering performance is to introduce a seprate `DataQuality` entity that keeps the relevant information. Any observation meeting the detail expressed in a `DataQuality` instance is then simply linked. This allows to filter on the `DataQuality` entities and then fetch the associated observations.

The other aspect that was identified is that `DataQuality` of a sensor may depend on time. So, the extension proposal for the data model includes another entity `HistoricalDataQuality`. This allows to produce a timeseries for data quality.

The following figure illustrates the proposal. The details of the `DataQuality` entity and the naming used is potentially misleading. But, because we want to contribute to a standardized solution and stimulate a good discusion, we share the following diagram

[image]

The `DataQuality` entity is linked to `Sensor` allowing the expression of information that can be applied to obaservations created by the sensor. The `DataQuality` entity also links to `Observation` and the STAplus `ObservationGroup` which allows to separate data quality for single observations and a collection of observations. To express the timeseries of data quality related to a sensor, the `HistoricalDataQuality` is linke to `Sensor`.

The diagramm and all illustrations above represent the current snapshot of thinking. It will for sure evolve over the next month...

humaidkidwai, 2024-02-02:
It's interesting to see DataQuality linked to Sensor, makes it easier to manage the maintenance of faulty Sensors. 

Would it not be more appropriate to have the DataQuality linked to Datastream and store the DataQuality attribute in Datastream? 

- one possible reason to do so would be to avoid redundancy, since all Observations in a Datastream are more likely to share certain attributes such as accuracy, precision, etc
- another reason could be the Quality of the data would be more viable to judge at an aggregate level rather individual Observations, some of which might be outliers.

The attributes for the DQ Entity do require more information

securedimensions, 2024-03-22:
First approach of a consolidated Data Model

[image]
STA version 1.1

[image]
STAplus version 1.1

schaefed, 2024-03-25:
The latest version of the consolidated model:

[image]
[image]

hylkevds, 2024-04-10:
For V2 we're renaming the `parameters` field to `properties`, so that should probably also be used here.
There seems to be a Typo in the diagram: HistoricalQuality**l**Feature

hylkevds, 2024-04-13:
Question about the cardinalities:
HistoricalQualityFeature can have multiple Sensors and multiple Observations. What does this mean?
I would expect a HistoricalQualityFeature to be the QualityFeature(s) a single Sensor or Observation had at a given point in time, so I would expect the HistoricalQualityFeature to have 1 Sensor _or_ 1 Observation.

hylkevds, 2024-04-13:
We may get questions about the term `QualityFeature`, and its overlap with the existing `Feature` class. Are there alternative names that could be considered?

nbrinckm, 2024-04-14:
Regarding the term: How about „QualityAnnotation“?

schaefed, 2024-05-30:
> I would expect a HistoricalQualityFeature to be the QualityFeature(s) a single Sensor or Observation had at a given point in time, so I would expect the HistoricalQualityFeature to have 1 Sensor or 1 Observation.

If we allow to 'reuse' a `QualityAnnotation` for several `Observations`/`Sensors` (which makes at least sense for categorical annotations) don't we then need the possibility to 'reuse' `HistoricalQualiotyAnnotations` as well? My use case would be a `Datastream`, whose `Observations` received two different `QualityAnnotation`s 'BAD' and 'OK'. At one point in time, we generate a new set of `QualityAnnotation`s for the same datastream and move all existing annotations to their historical counterpart. In order to realize such an example we need the given cardinalities, don't we?

> Also, how many QualityFeatures can an Observation have? The latest diagram has 0..1 but the original had 0..*.

I changed the cardinality from 0..* to 0..1 because of the introduction of `HistoricalQualityAnnotation`. The idea behind the changes was, that there should only be on valid quality information to a `Observation`/`Sensor` at any given point in time. But actually I am not sure anymore, as there might be compound `QualityAnnotations` as well (think of uncertainty ranges for example and distribution-like quality representations). Currently I think it would make sense to keep the 0..1 cardinality and allow for compound data types in `QualityAnnotation.result`, but I am actually unsure, if this makes sense. Any thought on this?

nbrinckm, 2024-06-07:
Stupid question from my side:

With regard to the new entity types that we want to add here - how do they related to the [Data Quality Vocabulary](https://www.w3.org/TR/vocab-dqv/)?
Can it be used or are there major problems?

I'm not really into the topic, got the hint from a colleague. If this is already part of your ideas, them I'm completely fine with this.

hylkevds, 2024-06-07:
> If we allow to 'reuse' a `QualityAnnotation` for several `Observations`/`Sensors` (which makes at least sense for categorical annotations) don't we then need the possibility to 'reuse' [...]

You mean you have a number of Observations in a Datastream, that are all linked to the `QualityAnnotation`s 'BAD' and 'OK'? And then you change the `QualityAnnotation`s for all those Observations to something else?
In one model we'd get a `HistoricalQualiotyAnnotation` per `Observation` / `Sensor`, in the other only one.

Both models would work. The question is which one is easier to implement for the client. Which, and how many POST/PATCH requests are needed, and what can be automated. If one changes each Observation to link to new `QualityAnnotation`s, and that automatically causes the creation of a new `HistoricalQualityAnnotation` with the current time, then we would end up with one-per, since each PATCH would have a different time. If one were to create a new `HistoricalQualityAnnotation`, with a fixed time, and all `Observations` linked, the server could check for each `Observation` if this is the newest `HistoricalQualityAnnotation`, and automatically re-link each `Observation` to the correct `QualityAnnotation`. In that case the many-per would work...
For HistoricalLocations both methods work.

If there is no automation in the background, any model would work, but that may be complex for the client...
 
> I changed the cardinality from 0..* to 0..1 because of the introduction of `HistoricalQualityAnnotation`. The idea behind the changes was, that there should only be on valid quality information to a [...]

Compound data types are harder for the client to deal with. And depending on the combinations that are possible, there may be very many different `QualityAnnotation`s... That would require some real-world testing to see which is better.

schaefed, 2024-06-11:
> With regard to the new entity types that we want to add here - how do they related to the [Data Quality Vocabulary](https://www.w3.org/TR/vocab-dqv/)?

Well, actually not. I am not an expert on the Data Quality Vocabulary (DQV) either but I read into it a bit and found, that we have the following, more or less congruent, definitions:
- `QualityMeasure` -> 'Quality Metric'
- `QualityAnnotation` -> 'Quality Measurement'
- `QualitySetup` -> 'Quality Annotation'

Furthermore DQV defines another entity the 'Quality Dimension', which acts as an abstract category for a number of 'Quality Metrics', that each cover a particular aspect of what it means for data to be considered "good" or "fit for purpose" (e.g. the Data 'Quality Dimension' Accuracy could cover 'Quality Metrics' that provide 'Quality Measurements' like error margins, absolute errors, z-scores, ...).

I am not sure, that we need something like the 'Quality Dimension' and I don't see much benefit in introducing another higher-level abstraction. If we would want do be compatible with / translatable into the DQV we likely should add the 'Quality Dimension' as well.

I like the terminology however and suggest the following renaming:
- `QualityMeasure` -> `QualityMetric`
- `QualityAnnotation` -> `QualityMeasurement` or, as the term 'measurement' is not used in STA at all, maybe `QualityObservation`?
- `QualitySetup` -> `QualityAnnotation` I am not sure if I really like the term `QualityAnnotation` here, but I dislike `QualitySetup` even more...

schaefed, 2024-06-11:
> You mean you have a number of Observations in a Datastream, that are all linked to the QualityAnnotations 'BAD' and 'OK'?    And then you change the QualityAnnotations for all those Observations to something else?
In one model we'd get a HistoricalQualiotyAnnotation per Observation / Sensor, in the other only one.

> Both models would work. The question is which one is easier to implement for the client. 

Then I would suggest to use the model, that is easier to implement and I leave the decision to you @hylkevds.

> Compound data types are harder for the client to deal with. 

I guess for the moment we could restrict the available types to scalars and fixed type arrays . If we stick with the `QualityAnnotation` as it currently is, then we likely need another `QualityAnnotation`-field, that allows us to distinguish the different annotations of a given `QualityMeasure` (e.g. we model an uncertainty range with two different `QualityAnnotation`s, one for the lower bound of the range, the other for the upper bound of the range, how could we tell them apart?). Only regarding usability compound data types would be more convenient, I think.

schaefed, 2024-06-13:
> Also, how many QualityFeatures can an Observation have? The latest diagram has 0..1 but the original had 0..*.

No matter how we decide on compound data types, we should allow multiple `QualityAnnotation`s for an `Observartion` in order to support hierarchical quality information.

anun333, 2026-10-08:
I have been building an open data and device contract for small farms — soil moisture, irrigation, livestock water — and the sensing side publishes through a SensorThings-compatible endpoint. Quality was the hardest part of our data model to get right, and in working through it we ended up answering several of the questions still open in this thread. Offering what we found, in case the evidence is useful.

Caveat up front: this is an implementation with tests, not a deployment with a field season behind it. Treat it as a worked example rather than as operational experience.

### On the `Datastream` approach proposed above — mapping instead of extending

We have one piece of hard evidence here, because we already use the parallel-datastream pattern for a different purpose: raw and calibrated values live in **separate immutable datastreams**, related to each other, rather than one overwriting the other. Recalibration appends a new series; it never rewrites history.

It works, and we would do it again. But it has a specific failure mode, and it is one the quality mapping would inherit. Our cross-check compares two sensors measuring the same property at the same feature. Once parallel datastreams existed, it began comparing a *raw* series against a *calibrated* one — `280` against `0.280` — and duly reported that the two probes violently disagreed. **A 99.8% false-alarm rate on a simulated farm with nothing wrong with it**, which we found only by scoring the detector against injected ground truth. The fix was to require that compared series agree on *kind and unit*, not merely on observed property and feature.

The lesson for this thread: if quality flags become `Observation`s in a sibling `Datastream`, then a generic client's most natural query — *every datastream for this observed property at this feature* — returns the measurements and the flags together, and nothing in the shape of the response says that one of them is not a measurement. The relation has to be machine-readable and the kind explicit, or clients will confidently compare the wrong pair. We read that as an argument for the mapping being **specified rather than conventional**, not as an argument against the mapping.

There is one question we could not answer for ourselves under that approach, on parameter provenance — below.

### On cardinality — 0..1 or 0..* annotations per Observation

We think there is a third option that dissolves the question. We first tried one annotation per check, and it loses information: when a flatline check and a range check both fire on the same reading, something has to choose which annotation is "the" current one, and the other check's finding silently disappears. That is a quiet failure of exactly the kind quality flags exist to prevent — we have a test that demonstrates it rather than asserting it.

What we settled on is **one annotation per assessment run, carrying a set of flags**. A run is "the suite of checks that looked at this reading, under this configuration"; its flags are the union of everything that fired. So an Observation has several annotations over time (re-assessments, corrections, human review) but exactly one *current* one, and multiple simultaneous findings live inside it rather than competing with each other. Cardinality stays 0..* over history and effectively 1 for "what does this reading mean right now", without needing a separate historical entity to make that distinction.

### On "compound data types are harder for the client"

That was our experience too, and we handled it by carrying both: the flag set, plus a single derived summary value (`good` / `questionable` / `bad`). A client that does not care about detail reads one scalar; a client that does reads the set. The important part is that the summary is *derived from* the flags by one published function, so two implementations cannot disagree about what a given set of flags means. If a standard specifies that mapping, the simple client gets portability for free.

### On storing the parameters a measure was configured with

Strongly agree this is needed, and it is where we would most want the standard to be prescriptive. We made thresholds **versioned records**, and each annotation names the version it used. The payoff is specific: when a farmer retunes a threshold, old annotations keep the old version, so a false alarm stays traceable to the rule that caused it and a retune cannot silently re-flag history. Without the version reference you can tell *that* something was flagged but not *under what rule* — and for us that distinction is what a certification auditor actually asks about.

This is the question we could not answer under the `Datastream` mapping. `ObservingProcedure` can carry the procedure, but a threshold version is not the procedure — it is the configuration the procedure ran with, and on a working farm it changes several times a season. Under the mapping, a retune seems to force a choice between forking a new `Datastream` per retune, which proliferates series for what is conceptually one assessment, and mutating the procedure in place, which destroys the ability to explain an old flag. If there is a third answer we have missed, that would settle a good deal for us.

### One thing I have not seen raised here: human override

[... cut ...]
Rather than a separate historical entity, we put a `supersedes` pointer on the annotation itself: a new annotation names the one it replaces, and the live one is whichever nothing supersedes. In practice this also answers the POST/PATCH question raised earlier in the thread — a re-assessment is one POST, the Observation is never touched, and the superseded annotation stays readable forever. Resolving "live" by set difference handles chains for free.

---

Happy to share the model in detail, or to implement a proposal from this thread against a working endpoint and report what breaks — that seemed to be what was wanted when compound types came up as "requiring some real-world testing".