opengeospatial/sensorthings #182: Representation of Observation metadata in the STA Model

doublebyte1, 2024-04-05:
We are using STA 1.1., to represent a series of neuroscience experiments. These experiments generate many datastreams with observations, but they also generate information which is not about the properties we want to measure, but about the actual measurements. They could be, for instance flags that tell us that a particular observation is important.

Does it make sense to represent these streams of metadata as `datastreams`? 

[image]

Looking at the data model, it seems datastreams require properties such as UnitofMeasurement which don't really apply to these more abstract properties. Are there alternative ways in which this information could be captured?

Thanks in advance for your guidance.

doublebyte1, 2024-04-05:
@glopesdev @PascalLike feel free to add your thoughts to the discussion.

humaidkidwai, 2024-04-05:
If the metadata is about each observation individually, the __parameters__ attribute within the Observation entity can be used to store such information. You can also just use the __properties__ attribute within the Datastream to store metadata for the whole collection of Observations

securedimensions, 2024-07-08:
Perhaps issue #174 has valuable ideas...