opengeospatial/sensorthings #192: Alignment with EDR Part 2: Publish-Subscribe Workflow

SpeckiJ, 2024-10-01:
The Amendment to the charter (22-034r1) states in Chapter 3.1:

> The OGC SensorThings API Part 1 - Sensing was published in 2016, and that was three years before the publication date of the OGCAPI - Features - Part 1:Core [17-069r3]. The SWG has done prelimenary work demonstrating that SensorThings API can be compatible with the OGC API. When the SWG start developing the sensing v2.0, the OGC API pattern will be considered in the design

I noticed that there has been ongoing work on aligning the normal REST Interface with EDR in https://github.com/opengeospatial/sensorthings/issues/135, but could not find any mention of `PubSub`-Alignment.

In the Connected-Systems SWG we were notified that apparently the EDR Part 2 is supposed to be the new basis for implementing PubSub in OGC APIs (https://github.com/opengeospatial/ogcapi-connected-systems/issues/68#issue-2521733062) and we should ensure that our specifications aligns to that. 

Is this also planned for STA despite not being mentioned in the update?

SpeckiJ, 2024-10-01:
Moved to mailinglist to allow for better discussion.

hylkevds, 2024-10-01:
Yes, we do need to look into OGC API Pub-Sub, to see if we can make that work.

And, since this is a good place to track this work, I'm re-opening this issue :)

hylkevds, 2024-10-01:
One potential issue with AsyncAPI is the same as with OpenAPI: STA has a near-infinite number of potential topics, leading to massive API documents. We'll have to do a "most useful" list to make the document size realistic.