opengeospatial/sensorthings #198: Example of

liangsteve, 2025-03-04:
We need some examples of proximateFeatureOfInterest and ultimateFeatureOfInterest relations in v2.0

For example (Listing 19) seems incorrect to me (```  "proximateFeatureOfInterest@navigationLink": "Observations(1)/proximateFeatureOfInterest"```)

```json
{
  "@context": "https://example.org/v2.0/$metadata#Observations/$entity",
  "@id": "Observations(1)",
  "id": 1,
  "phenomenonTime": {
    "start": "2017-11-12T13:00:00Z",
    "end": "2017-11-12T14:00:00Z"
  },
  "resultTime": "2017-11-12T14:00:00Z",
  "result": 12.5,
  "Datastream@navigationLink": "Observations(1)/Datastream",
  "proximateFeatureOfInterest@navigationLink": "Observations(1)/proximateFeatureOfInterest"
}
```

Should it be `"proximateFeatureOfInterest@navigationLink":"Observations(1)/Feature"`?

@humaidkidwai Thanks!  :)

hylkevds, 2025-03-04:
No, the relation is called `ProximateFeatureOfInterest`, so the link becomes `Observations(1)/ProximateFeatureOfInterest`.
The P should be capitalised though, so the example is indeed incorrect :)

A data model may have multiple links to the same entityType, and when the link would use `SourceType(id)/TargetType`, the server would not know which link is meant. 
For example, in the [relations extension](https://hylkevds.github.io/24-046/24-046.html#relations-extension), each relation has two links to the same Type: `source` and `target`. So theirs links are:
- RelatedDatastream(1)/Source
- RelatedDatastream(1)/Target
And both links point to a Datastream.

hylkevds, 2025-03-05:
The [Feature](https://hylkevds.github.io/23-019/23-019.html#feature) examples have the links to Datastreams and Observations. But the [Datastream](https://hylkevds.github.io/23-019/23-019.html#datastream) examples are missing all navigation links.

I've fixed the `p`.

hylkevds, 2025-03-05:
Though technically those Datasteam examples are correct, since the caption doesn't state they are as _returned by a HTTP end point_ like the other examples do. We may want to keep those Datastream examples as is, since the focus of those examples is the resultType, and the navigation links and selfLink would just add more distraction...

humaidkidwai, 2025-03-06:
The navigation links in the examples of [Feature ](https://hylkevds.github.io/23-019/23-019.html#feature) entity are incorrect. They use the older terminology - `FeaturesOfInterest(id)/relatedEntity`

hylkevds, 2025-03-07:
> The navigation links in the examples of [Feature ](https://hylkevds.github.io/23-019/23-019.html#feature) entity are incorrect. They use the older terminology - `FeaturesOfInterest(id)/relatedEntity`

Fixed it :)