opengeospatial/sensorthings #200: Spec 2.0 progress and branches

matthijskooijman, 2025-04-01:
I'm considering to use the sensorthings spec for a project, and I notice there is work underway for a 2.0 version that looks interesting.

 I have two questions:
 - What is the status/progress/timeline? The charter mentions the plan to submit for approval in 2022, but I'm thinking that is no longer accurate.
 - What is the meaning of the various versions / branches? I see 15-078 / 17-079 / 18-088 / 22-022 are the published sensing 1.0 / tasking 1.0 / sensing 1.1 and staplus specs, but I see recent changes in the 23-019, 24-032 and 24-046 branches. 24-032 seems to be a separate add-on spec for Websub, but both 23-019 and 24-046 claim to be the the SensorThings API 2.0. It seems 23-019 has seen the most changes (last from last month), while 24-046 started off as a copy of 23-019 last August (but omits the core sections). Maybe 24-046 is also intended as a separate spec for some extensions, but just did not have its title updated?

humaidkidwai, 2025-04-02:
The expected timeline to submit the revised spec is not a fixed date. However, the SWG is trying to finalize the draft before the next OGC member meeting scheduled in June. The charter is not updated to reflect the same as of now.

23-019 is the doc no of the main spec, whereas 24-046 is an extension spec that specifies the official data model extensions of the standard for complex observational systems. 24-046 is planned to be submitted later once 23-019 is finalized.

hylkevds, 2025-04-02:
I've fixed the title of 24-046.

matthijskooijman, 2025-04-03:
Thanks for clarifying. I'll submit a PR to add branch info to the README.

> I've fixed the title of 24-046.

I do not see that here, did you forget to push? https://github.com/opengeospatial/sensorthings/commits/24-046/

hylkevds, 2025-04-03:
Yep, forgot to push :)

PR #202 (2025-04-03): Document branches in README
This fixes #200

The titles in the list are taken from the document titles, except for 24-046 (which has not had its title updated, so I based it on the README title, slightly modified to be more consistent with the rest).
commit e77b0b6b (2025-04-03): Document branches in README

This fixes #200
closed 2025-04-03 by hylkevds